ayushsharma82/AsyncElegantOTA #1: Attempting to compile for ESP8266 and get no matching function call error

hewfish, 2019-06-14:
Using latest master versions of

ESPAsyncTCP, ESPAsyncWebServer and AsyncElevantOTA. Trying to compile the ESP8266 example and i get the following errors:

C:\Users\hewfish\Desktop\nodeMCU\test\asyncupdate\asyncupdate.ino: In function 'void setup()':

asyncupdate:34:31: error: no matching function for call to 'AsyncElegantOtaClass::begin(AsyncWebServer&)'

   AsyncElegantOTA.begin(server);    // Start ElegantOTA

                               ^

C:\Users\hewfish\Desktop\nodeMCU\test\asyncupdate\asyncupdate.ino:34:31: note: candidate is:

In file included from C:\Users\hewfish\Desktop\nodeMCU\test\asyncupdate\asyncupdate.ino:5:0:

C:\Users\hewfish\Documents\Arduino\libraries\AsyncElegantOTA\src/AsyncElegantOTA.h:28:14: note: void AsyncElegantOtaClass::begin(AsyncWebServer*)

         void begin(AsyncWebServer *server){

              ^

C:\Users\hewfish\Documents\Arduino\libraries\AsyncElegantOTA\src/AsyncElegantOTA.h:28:14: note:   no known conversion for argument 1 from 'AsyncWebServer' to 'AsyncWebServer*'

Multiple libraries were found for "AsyncElegantOTA.h"
 Used: C:\Users\hewfish\Documents\Arduino\libraries\AsyncElegantOTA
 Not used: C:\Users\hewfish\Desktop\Arduino-NodeMCU\arduino-1.8.9-windows\arduino-1.8.9\libraries\AsyncElegantOTA
exit status 1
no matching function for call to 'AsyncElegantOtaClass::begin(AsyncWebServer&)'

ayushsharma82, 2019-06-14:
It should be `AsyncElegantOTA.begin(&server);` instead of `AsyncElegantOTA.begin(server);`

Please compile again with the above change in your code and let me know.

hewfish, 2019-06-14:
That did the trick, thanks

closed 2019-06-14 by hewfish