ayushsharma82/AsyncElegantOTA #2: Suggestion for update to AsyncElegantOTA.h and elegantWebpage.h to simplify changes

hewfish, 2019-06-22:
Love the library. Just a small suggestion to make making changes to the elegantWebpage.h simpler. Rather than hard-coding ELEGANT_HTML_SIZE into elegantWebpage.h. Remove that const completely and compute it on the fly.

This would only require a simple change in AsyncElegantOTA.h

AsyncWebServerResponse *response = request->beginResponse_P(200, "text/html", ELEGANT_HTML, ELEGANT_HTML_SIZE);

becomes

AsyncWebServerResponse *response = request->beginResponse_P(200, "text/html", ELEGANT_HTML, **sizeof(ELEGANT_HTML)**);

Then no matter how the PROGMEM array changes in the elegantWebpage, it will always load correctly without having to recompute the uint32 size and hardcoding it.

ayushsharma82, 2019-12-16:
It has to do with the nodejs script which I created for converting the webpage into a PROGMEM array. I will change it in future. Thanks for your suggestion 👍

fullspamfullspam, 2019-12-19:
no, it hasn't. The ELEGANT_HTML array can be the same so you don't have to convert anything.
Hewfish suggest to change size const for a function related to ELEGANT_HTML (sizeof) for futures updates

ayushsharma82, 2019-12-19:
@fullspamfullspam Yes i understand that 😅 But there is a whole template for this layout in that script. Please refer to ESP-DASH to see finalize.js.

closed 2019-12-19 by ayushsharma82