ayushsharma82/AsyncElegantOTA #8: Compile error using current ESP-IDF for ESP32: Hash.h: No such file or directory

ul-gh, 2019-10-30:
Hi,

first, thanks for your nice contribution with this software!

However, the current version ESP-IDF for ESP32 does not contain a "Hash.h" header any more,
compilation fails with "AsyncElegantOTA.h:13:22: fatal error: Hash.h: No such file or directory"

However, the include does not seem to be necessary,
removing the #include "Hash.h" from line no. 13 makes the compilation succeed.

finchamp, 2019-11-19:
The same issue exists in the current version of Arduino. The same solution applies.

ayushsharma82, 2019-12-16:
Great. I will test this change and create a pull request.

closed 2020-01-27 by ayushsharma82

JoeBarouneD, 2020-01-28:
Hi, so could you remove the Hash includes in the examples (and in readme) ?

smford, 2020-06-25:
This problem still exists in the examples

ayushsharma82, 2020-06-25:
@smford 
With v2.2.1 release, I have added a copy of Hash.h in the `src` folder itself which has fixed that 'missing' error.

Let me know if it still occurs.  Thanks!

ayushsharma82, 2020-06-25:
@JoeBarouneD  Yeah, I'll do it now, since it is now included in the library itself.

ajinkya-open, 2023-01-05:
i faced this issue but when i #include<FS.h> its compiling well 

here is snippets

#include <Arduino.h>
#if defined(ESP8266)
#include <ESP8266WiFi.h>
#elif defined(ESP32)
#include <WiFi.h>
#endif

#include <Ticker.h>
#include <ESPAsyncWebServer.h>
#include "fauxmoESP.h"
#include <DNSServer.h>
#include <PubSubClient.h>
#include <AsyncMqtt_Generic.h>
 #include <FS.h> // fs added and has.h error gone
#include <ArduinoJson.h>
#include "config.h"

#include "SPIFFS.h"
extern "C"
{
#include "freertos/FreeRTOS.h"
#include "freertos/timers.h"
}