ayushsharma82/AsyncElegantOTA #9: Complie error

giangi007, 2019-11-09:
Hello i receive this error when I try to compile your example for esp32:
Scheda:"WEMOS LOLIN32, 80MHz, Default, 240MHz (WiFi/BT), 115200"

C:\Users\giang\OneDrive\Documenti\Arduino\libraries\Hash\src\Hash.cpp:26:34: fatal error: bearssl/bearssl_hash.h: No such file or directory

compilation terminated.

Could you help me?

thanks

skippa, 2019-11-20:
Same issue here, Hash.h isn't found and is throwing compile error... I googled around and all I seem to find is references to PlatformIO issues that throw the same error.

How can we find and include the Hash.h lib?

ayushsharma82, 2019-12-16:
Please refer to https://github.com/me-no-dev/ESPAsyncWebServer/issues/46

closed 2019-12-16 by ayushsharma82