ayushsharma82/AsyncElegantOTA #14: Not compiling with latest Arduino IDE

Vrumfondel, 2020-02-01:
The Project is currently not compiling.
Due to my analysis so far it seems the compiler is not able to handle the definition of ELEGANT_HTML[] in one single line, leading to quite bogus compiler errors.

Vrumfondel, 2020-02-01:
OK, seems a clear Arduino Pre-Processor issue.
Whenever AsyncElegantOTA.h is added - the preprocessor is not generating any function declarations any more.

Vrumfondel, 2020-02-01:
OK,
nailed it down.
The preprocessor hat problems handling ELEGANT_HTML[] if defined in one single line. If you simply add LF's in between its good, so please do so.

chriswant, 2020-02-04:
Hello,
I'm having the same issue.  And getting Error: " int cmd = (filename.indexOf("spiffs") > -1) ? U_FS : U_FLASH;"...     
Problem points to  ELEGANT_HTML[]...
How do I add the LF's to make it work?
Thanks

Vrumfondel, 2020-02-04:
Hi!
The issue with U_FS is solved when you update to the latest version (w.as updates yesterday in the arduino libs)

Adding LF's: Open the .h in an editor of your choice, and make the one line into multiple ones.
Or wait until the author changes this. I'm going to create an pullrequest for this later ..

chriswant, 2020-02-04:
Thank you for your fast reply. 
 I have done what you said and I still get the error: 'U_FS' was not declared in this scope"...
All of My Libraries are up to date.
What else could be wrong?
Using just the smple ESP8266_Async_Demo.ino
Thank You

Vrumfondel, 2020-02-07:
So you are on ESP8266 and not ESP32?
Which board did you select?

chriswant, 2020-02-07:
Hello,
The board is an "AI ESP8266 Node MCU ESP-12E".
Board Selected is  "NodeMCU 1.0 (ESP-12E Module).
The library is AsyncElegantOTA 1.0.6 Updated.
Still the same error...

In file included from AsyncElegantOTA\examples\ESP8266_Async_Demo\ESP8266_Async_Demo.ino:5:0:
AsyncElegantOTA\src/AsyncElegantOTA.h: In lambda function:
AsyncElegantOTA\src/AsyncElegantOTA.h:51:71: error: 'U_FS' was not declared in this scope
                         int cmd = (filename.indexOf("spiffs") > -1) ? U_FS : U_FLASH;

Thank You for your Help.
                                                                       ^

Vrumfondel, 2020-02-07:
Hm. Building for me with this board. There must be something wrong with your Arduino IDE installation. In the NodeMCU Board packede U_FS ist definitly defined.
How did you install the package fo 8266 boards?

chriswant, 2020-02-09:
Hello,
Im using these Librariers:
Using library ESP8266WiFi at version 1.0 in folder: \esp8266\hardware\esp8266\2.5.2\libraries\ESP8266WiFi 
Using library Hash at version 1.0 in folder: 
\esp8266\hardware\esp8266\2.5.2\libraries\Hash 
Using library ESPAsyncTCP-master at version 1.2.0 in folder: 
\Arduino\libraries\ESPAsyncTCP-master 
Using library ESPAsyncWebServer-master at version 1.2.2 in folder: 
\Arduino\libraries\ESPAsyncWebServer-master 
Using library AsyncElegantOTA at version 1.0.6 in folder: 
\Arduino\libraries\AsyncElegantOTA

Using Package:
https://arduino.esp8266.com/stable/package_esp8266com_index.json

Installed via GIT.

chriswant, 2020-02-09:
Hello,
I have updated the ESP8266 Board in Board Manager to 2.6.3.
I have now successfully updated my ESP8266 Node MCU with ESP8266_Async_Demo.ino.
Thank You again for your Time.

chriswant, 2020-02-09:
Hello,
So I was able to install ESP8266_Async_Demo.ino on to the ESP8266 but when
I look for the SSID to logon on to the ESP8266 its not there and the Serial
monitor shows the dots continuously.  What could I be doing wrong.\

Thank You
Chris

sgf379, 2020-02-09:
Hi chriswant,

Have entered your WiFi SSID and WiFi password in to the following assignments in the sketch:

const char* ssid = "........"; const char* password = "........";

If so, check that they are entered correctly.

Regards
Steve

chriswant, 2020-02-10:
Thank you again.  I got it working now.  I was wondering if there is a way
to make to ESP8266 Node have its own SoftAP to upload OTA Directly without
a Wifi Router?

Thank You
Chris

closed 2020-11-06 by ayushsharma82