ayushsharma82/AsyncElegantOTA #16: OK on ESP8266 but not on ESP32

roberttidey, 2020-02-06:
I can't get this OTA to work on ESP32. Compiles OK and interface comes up but upload stalls and fails after about 10% progress and serial shows an abort is called on core 1.

Arduino environment 3MB/1MB flash split. Compiled binary in 1.1MB

So I thought I'd try using the example code. First issue I got was missing Hash.h. Tried commenting that include out and compiled ok, but behaved as before with abort and stall. I don't know whether missing Hash.h out could be associated with that but I would have expected a compile error of it was actually being used.

I have checked my ESP8266 and ESP32 environments. ESP8266 does indeed have the Hash library but ESP32 does not.

I have also verified that this does run fine on my ESP8266 set up.

Tigger777, 2020-02-07:
Hi Robert,

I've been playing with this this morning and have had no issues.
Have you tried it with the default settings?

roberttidey, 2020-02-07:
Thanks for checking.

For my test with example code all I did was change the ssid and password and then commented out the Hash.h to allow it to compile. Serial debug then shows abort issued very soon after upload starts.

I then added a bit of debug into AsyncElegantOTA.h 
declared 
size_t written_len;

then replaced chunk write with

written_len = Update.write(data, len);
Serial.println("upi:" + String(index) + " upl:" + String(len) + " upw:" + String(written_len));
if (written_len != len) {
   Update.printError(Serial); 
 }

This shows first 2 chunks get written but then the abort is triggered. So it looks like something is going wrong with Update.write  I see that the esp8266 version has an Update.runAsync call which doesn't exist in ESP32. 

I am using ESPAsyncWebServer v1.2.3 and compiling under Arduino using the standard Espressif ESP32 package. That dates from Sep 2019; I could try updating that to see if that makes any difference.

Tigger777, 2020-02-07:
Hi Robert,

I was a bit vague this morning, tried attach a screen shot which did't work.
I downloaded it this morning, and my libs are all up to date. 
I used the default partition scheme 4M + SPIFFS 1.2mb App + 1.5mb Spiffs on Arduino version 1.8.10.

Here, https://esp32.com/viewtopic.php?t=6954, is a mention of the partitions on the ROM.
nvs,      data, nvs,     0x9000,  0x4000
otadata,  data, ota,     0xd000,  0x2000
phy_init, data, phy,     0xf000,  0x1000,
factory,  app,  factory, 0x10000, 2M,
storage,  data, spiffs,  ,        0xF0000,
ota_0,    0, ota_0,     0x400000, 2M,
ota_1,    0, ota_1,     0x600000, 2M,

You need to have 2 OTA partitions, as the one will write to the other when updating and then set the boot partition and reboot.

Hope this helps
Duane

roberttidey, 2020-02-07:
OK. That helped a lot and I now have it working fine.

I was using the ESP32Cam set up and this didn't have a partition scheme menu. When I checked it had a fixed scheme to 3MB App no OTA.

I added in the partitions scheme menu and it worked straight away.

I'll close this issue.

closed 2020-02-07 by roberttidey