ayushsharma82/AsyncElegantOTA #21: Upload shows completed, but still running old firmware

Derek-K, 2020-03-29:
I have spent about good 30 minutes keep recompiling and doing OTA update, turns out AsyncElegantOTA was broken.

Upload shows completed on the webpage. The first issue, ESP8266-01 doesn't reboot. No big deal, I just recycle power. But turns out, it is still booting up with old firmware the whole time!!

I have tried to wait for a few minutes after the upload is completed before recycling power, the same result, only booting with old firmware...

A few references point:
HW: ESP8266-01 with 1MB
CPU: 80MHz
Flash size: 1MB (FS:64KB OTA ~470KB)
Sketch uses 434748 bytes
Actual bin file size 438896 bytes

sgf379, 2020-03-29:
> I have spent about good 30 minutes keep recompiling and doing OTA update, turns out AsyncElegantOTA was broken.
>
> Upload shows completed on the webpage. The first issue, ESP8266-01 doesn't reboot. No [...]

Hi Derek-K,
I have been using AsyncElegantOTA for the last 6 months on several projects without issue. The only time I have seen the behaviour you are describing, is when the firmware being uploaded is the same as the firmware on the device. 
Regards
Steve-F

ayushsharma82, 2020-03-29:
Hi @Derek-K ,
I remember when I got this error. It was basically due to the HTTP request `content-length` header being larger than the size of the actual bin file. ( For example, your bin file is 438Kb, the HTTP request will show 441Kb which is larger than the Flash config for your ESP8266. Therefore your module doesn't accept the firmware you are trying to upload )

This is not an issue with your bin file / flash config, both are working perfectly. The AsyncElegantOTA library uses `Content-Length` header for HTTP request which sometimes cannot show the size of the bin file precisely. 

I'll be pushing an update soon with the v2 user interface ( from ElegantOTA repo ) which shows errors in a better way and also a fix for this issue.

Kudos! for making this come to my mind.

ayushsharma82, 2020-03-29:
@Derek-K  Please check the (https://github.com/ayushsharma82/AsyncElegantOTA/tree/v2)[V2] branch for the updated code. 

Let me know if that fixes your problems.

Regards,
Ayush

Derek-K, 2020-03-29:
> I have been using AsyncElegantOTA for the last 6 months on several projects without issue. The only time I have seen the behaviour you are describing, is when the firmware being uploaded is the same as the firmware on the device.

LOL. I am sure that would happen. 

But no, I am sure it is a different firmware because I changed the display string, recompiled, checked the new file size, and creation date and time.

Derek-K, 2020-03-29:
> @Derek-K Please check the [V2](https://github.com/ayushsharma82/AsyncElegantOTA/tree/v2) branch for the updated code.
> 
> Let me know if that fixes your problems.
> 
> Regards,
> Ayush

Cool! I will give it another try later... (Sorry, I switched to my old OTA library after discovering this issue. But I will try to pull my saved version that uses AsyncElegantOTA, update to V2 and try again later)

ayushsharma82, 2020-06-24:
Fixed with the release of [V2.2.0](https://github.com/ayushsharma82/AsyncElegantOTA/releases/tag/v2.2.0). 

Now starts with the maximum allowed size for an ESP8266 which fixed this issue.

closed 2020-06-24 by ayushsharma82