ayushsharma82/AsyncElegantOTA #23: ESP 32 does not compile

chmasi, 2020-04-15:
Hi 

I try to use your Library but it will not compile.

I will get this errormessage

no known conversion for argument 1 from AsyncWebServer to AsyncWebServer*

any suggestion?

Martin

barthogenes, 2020-04-21:
Hello Martin,

I had the same issue when calling `AsyncElegantOTA.begin(server);`.
The function signature is `void begin(AsyncWebServer *server)` which means that it expects a pointer instead of a copy of the AsyncWebServer variable.
So to start the web server use:
```
AsyncElegantOTA.begin(&server);
```
instead of:
```
AsyncElegantOTA.begin(server);
```

chmasi, 2020-04-21:
Thanks a lot,
Freundliche Grüsse
Martin Sickel
______________________________________________

Oberdorfstr. 28
78337 Öhningen
+49 7735 3551
martin@die-sickels.de

closed 2020-06-22 by ayushsharma82