ayushsharma82/AsyncElegantOTA #26: compile error in file AsyncElegantOTA.h from 24Jun2020

HaegarDerWikinger, 2020-06-25:
Hi Ayush,
please could you check this out in the new version?
Thank You and Best Regards,
Michael

C:\Users\Michael\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.6.3\libraries\AsyncElegantOTA/AsyncElegantOTA.h:94:52: error: '_FS_end' was not declared in this scope

                         size_t fsSize = ((size_t) &_FS_end - (size_t) &_FS_start);
same for _FS_start

ayushsharma82, 2020-06-25:
I'll setup CI in a minute. It will provide better tests for future commits.

ayushsharma82, 2020-06-25:
Fixed with v2.2.1 release.

Thanks for pointing that out!

closed 2020-06-25 by ayushsharma82

smford, 2020-06-25:
> Fixed with v2.2.1 release.
> 
> Thanks for pointing that out!

Is that a version typo - the release that had the issue was v2.2.2, and you've made a new release v2.2.1 - were you meaning v2.2.3? 

I got it working in v2.2.1

ayushsharma82, 2020-06-25:
It was `v2.2.0` which had that issue.

Now the latest release tag is `v2.2.2`.
I got confused as well 😆

iambappa, 2020-07-15:
Still get below error when compiled for ESP8266

C:\Users\mypc\Documents\Arduino\libraries\AsyncElegantOTA\src/AsyncElegantOTA.h:95:64: error: 'U_FS' was not declared in this scope

                         int cmd = (filename == "filesystem") ? U_FS : U_FLASH;

ESP32 compiles fine.

ayushsharma82, 2020-07-15:
@iambappa Try updating your ESP8266 Arduino Core to the latest release.

iambappa, 2020-07-16:
Still same error after updating to latest release