ayushsharma82/AsyncElegantOTA #28: [How to] Automatically upload a bin firmware file?

pranjal-joshi, 2020-06-30:
Hello,
For my project, I am using an android app to download a `.bin` update file from a remote server and I want to upload it to the ESP8266 via my custom android App.
Can you please share what is `GET` or `POST` method so that I can programmatically push the firmware file instead of using WebBrowser UI?

Thanks for the awesome library!

ayushsharma82, 2020-07-01:
Hi,
You can do a `POST` request to the `/update` URL.
While making the request, Append the form data with the file name as 'filesystem'  or 'firmware' according to your needs.

Your module's web server will return an HTTP code based on if the OTA was successful or not.

I hope this information was helpful to you. Thanks for using AsyncElegantOTA!

pranjal-joshi, 2020-07-01:
Ok, Great!
For example, I am attaching a sample HTML code so that u can tell me whether the parameters are correct or not?
```
<form action="/update">
  <input type="file" id="myCustomUpdateFile" name="firmware">
  <input type="submit">
</form>
```
Let me know :)

ayushsharma82, 2020-07-03:
@pranjal-joshi  I think so. It may depend on the actual filename as well. Ex. `firmware.bin`. In my UI, the javascript will automatically overwrite the name before sending it to the ESP.

pranjal-joshi, 2020-07-03:
> @pranjal-joshi I think so. It may depend on the actual filename as well. Ex. `firmware.bin`. In my UI, the javascript will automatically overwrite the name before sending it to the ESP.

Yes. The actual file name will be like `something.nodemcu_firmware.bin`
So do I have to rename the file? Or I have to use `"name"` attribute like `"name"="firmware"` irrespective of actual filename?

pranjal-joshi, 2020-07-05:
@ayushsharma82 I have tried following code to get started but it's not working.
Can u help me fixing the problem?
```
<form action="http://192.168.4.1/update" method="POST">
Choose a file to upload: <input name="firmware" value="firmware" type="file" /><br />
<input type="submit" value="Upload File" />
</form>
```

ayushsharma82, 2020-07-05:
Try using ElegantOTA instead. This one is made specifically to work with some javascript interference.

pranjal-joshi, 2020-07-06:
> Try using ElegantOTA instead. This one is made specifically to work with some javascript interference.

I went through [ElegantOTA/ui](https://github.com/ayushsharma82/ElegantOTA/tree/master/ui) code but it seems kinda similar with the AsyncElegantOTA!! As I am new with ESP8266, I might not be getting your point clearly. Can you please guide me in detail or send a documentation link that explains how to properly create `POST` request?

ayushsharma82, 2020-07-09:
It is out of the scope of this repository, but I'll point you to the right direction:

You can use the HTML provided in this file: https://github.com/esp8266/Arduino/blob/master/libraries/ESP8266HTTPUpdateServer/src/ESP8266HTTPUpdateServer-impl.h#L22

closed 2020-07-15 by ayushsharma82

mindrunner, 2021-09-17:
I am using:

```
file=/path/to/firmware.bin
md5=($(md5sum $file))
mode=firmware

curl --digest -u $user:$password --compressed -L -X POST -F "MD5=$md5" -F "name=$mode" -F "data=@$file;filename=$mode" "$host/update"
```

to automatically update boards.