ayushsharma82/AsyncElegantOTA #30: more a question than an issue

simonjamain, 2020-07-30:
Hi,
I use SPIFFS.h on my esp32 project (https://github.com/simonjamain/openmosfet) and I need to upload files as well as the software (in OTA), I'm struggling to find any usefull information on how to do it.

Does your library allows it ?
How do i do it ?
How do I create a .bin of my files (assuming that it is needed) ?

Thank you very much, great piece of software BTW.

ayushsharma82, 2020-07-30:
Yup, This library allows you to update both firmware and filesystem.

You can get the spiffs image ( .bin ) path from the logs shown in Arduino IDE. 

1. Don't connect your ESP32 to your PC
2. Click 'Upload sketch data'
3. File path will be shown in the logs of Arduino IDE. ( it would be something like APPDATA / spiffs.bin etc. )

simonjamain, 2020-07-30:
noice thank you very much, I'll try that !

closed 2020-07-30 by simonjamain