ayushsharma82/AsyncElegantOTA #31: Arduino lib builder abort() on core 0

handmade0octopus, 2020-08-16:
Standard ESP32 example on standard Arduino IDE with default file system etc.

Can't update over OTA, getting "[HTTP ERROR] Bad Request"

Serial back log.

```
HTTP server started
assertion "new_rcv_ann_wnd <= 0xffff" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/core/tcp.c", line 779, function: tcp_update_rcv_ann_wnd
abort() was called at PC 0x40105a17 on core 0

Backtrace: 0x4008c434:0x3ffb3d40 0x4008c665:0x3ffb3d60 0x40105a17:0x3ffb3d80 0x4011da22:0x3ffb3db0 0x4011dab1:0x3ffb3dd0 0x400d284e:0x3ffb3df0 0x40118ee0:0x3ffb3e10 0x40088b7d:0x3ffb3e40

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:9720
ho 0 tail 12 room 4
load:0x40080400,len:6352
entry 0x400806b8
```

ayushsharma82, 2020-08-16:
Hi, 
Which version of this library are you using?

ayushsharma82, 2020-08-16:
@handmade0octopus  This should be fixed in the latest release: [v2.2.4](https://github.com/ayushsharma82/AsyncElegantOTA/releases/tag/v2.2.4)

Please do try it out and let me know. Thanks for using AsyncElegantOTA.

handmade0octopus, 2020-08-17:
Working, thanks!

closed 2020-08-17 by handmade0octopus