ayushsharma82/AsyncElegantOTA #32: Update is failing for nodemcu esp8266

hey24sheep, 2020-08-16:
I am getting this error when I am trying to upload .bin using this library.

Board : NodeMCU ESP8266

In prettified js file
`Failed to load resource: net::ERR_CONNECTION_RESET`

In console
`POST http://192.168.1.50:8080/update net::ERR_CONNECTION_RESET
(anonymous) @ update:formatted:5150
Promise.then (async)
uploadOTA @ update:formatted:5146
re @ update:formatted:1183
r @ update:formatted:1313
Ko.i._wrapper @ update:formatted:3450`

ayushsharma82, 2020-08-16:
@hey24sheep  Can you check if the firmware was updated on reboot or it stayed same?

ayushsharma82, 2020-08-16:
I think the correct update URL should be `http://192.168.1.50/update`. ( without the 8080 port )

hey24sheep, 2020-08-16:
I'm using custom port 8080 as I have fauxmo running on port 80 and I cannot change that. Your webpage gets loaded without any issues. It goes to 55 to 67 percent but then the connection resets.

ayushsharma82, 2020-08-16:
I see. Can you paste serial monitor log?

hey24sheep, 2020-08-18:
Hey, what kind of log do you need? I don't see any logs from your library? Is there any debug flag or something?

ayushsharma82, 2020-08-18:
Okay. I'll check it in a day or two on my own esp8266.

hey24sheep, 2020-08-18:
@ayushsharma82 It was heap issue. It was getting out of memory or something and it was breaking up the connection as esp was restarting. So, it was giving connection error. I removed a lot of useless code from my sketch and it started working fine. But, it will still fail if big sketch is uploaded and during update it doesn't have enough mem.

hey24sheep, 2020-08-18:
I would say one idea, send heap data with your progress on UI it will be helpful.

ayushsharma82, 2020-08-19:
Great! Have fun. 

Closing this issue.

closed 2020-08-19 by ayushsharma82