ayushsharma82/AsyncElegantOTA #33: OTA fails with access point

djswirl, 2020-08-18:
I've been testing your library, I need to do an update via a browser when running in access point but it fails.
Your example code work fine if I connect to a wifi router but if I set up an access point it gets to about 40% then hangs.
Is there anyway to run this via an access point 

Many thanks

ayushsharma82, 2020-08-18:
I'll test my library in this scenario and let you know.

Juand0805, 2020-09-13:
Hi @ayushsharma82 thank you for this excellent work, is working fine accessing via wifi router, now im trying to use it and adapt the library for an ESP32 working as SofAP, but there's no answer when accessing to the elegant web page. are you using any Ajax inside it. do you know how this can be solved if I have a jquery embedded on the code ?

blaniosh, 2020-10-02:
When I attempt to update firmware the update gets to 9% and then stops. Access is via Wireless access point. If I connect directly to my router (which is out of range of my module), the update completes OK.

CODeRUS, 2020-11-06:
not working with WIFI_AP mode

djswirl, 2020-11-06:
After doing some research after my first post it turns out OTA does not work in AP mode. That's not just for this library it's for any code, so far I've not found any solution

ayushsharma82, 2020-11-06:
Guys, Currently I'm not getting any time look into this issue, till then if anybody finds a solution please be free to open a pull request. 

More info requested from debuggers:
- Which ESP are you using?
- What's your flash config?
- Please paste the exception code here if it's shown in your Serial monitor.

CODeRUS, 2020-11-06:
> 
> 
> After doing some research after my first post it turns out OTA does not work in AP mode. That's not just for this library it's for any code, so far I've not found any solution

what makes you to think like that? I am using OTA in STA mode with no problem in my small project: https://github.com/CODeRUS/vehicle-lights

djswirl, 2020-11-06:
Exactly it works in STA mode, but not in AP mode

CODeRUS, 2020-11-06:
> 
> 
> Exactly it works in STA mode, but not in AP mode

In AP mode, did you spent your time to check my code? :)

djswirl, 2020-11-07:
No. But I've yet to find any code that works in AP mode. If you are saying your code works in AP I'll check it out :)

Sejny, 2021-02-07:
@CODeRUS   Feeding wdt in your code maybe solve problems with updating over AP [https://github.com/CODeRUS/vehicle-lights/blob/058093d3bee88accf24fcbb145187e8711c1cdfd/src/main.cpp#L152](url)

#45 With this modification update work in AP mode, STA mode and also over ethernet. I belive, this solution is not clean.

CODeRUS, 2021-02-07:
Thanks but i've switched from cool but stale projects to self-coded libraries :)

ayushsharma82, 2021-02-07:
_I was browsing through the convo in my mail, so here my thoughts on that comment:_

@CODeRUS  Or you could have just written a single line like @Sejny did to fix the whole thing for everyone

Open-source is for collaboration rather than depending on the author to fix the whole thing :)

closed 2021-12-18 by ayushsharma82

hungyip, 2022-03-17:
I am also having this issue. Tried to update throught ElegantOTA but it always got stuck in around the middle. I only use AP mode for my project.