ayushsharma82/AsyncElegantOTA #34: How to compile/build own "elegantWebpage.h"

IAmOrion, 2020-09-10:
I download ElegantOTA as the readme says the source is there.... 
I'm on MacOSX - the only part that refuses to run is 'npm run lint' - which I'm hoping isn't the final build part as no matter what I do, it always fails saying vue-cli-service command not found despite me adding required PATH variables and uninstalling and reinstalling vue/cli 10 times!

How to I 'compile' the source and create the .h file?

ayushsharma82, 2020-09-10:
You'll have to install Nodejs first: https://nodejs.org/en/
Then only those `npm` and `node` commands will be available for your use.

ayushsharma82, 2020-09-10:
For compiling the whole project into elegantWebpage.h just execute `npm run build`. It will automatically replace the existing elegantWebpage.h with the newer one.

IAmOrion, 2020-09-10:
> You'll have to install Nodejs first: https://nodejs.org/en/
> Then only those `npm` and `node` commands will be available for your use.

I have already installed NodeJS & NPM and vue/cli but the npm run lint still fails.

npm run build DOES work however, although it just creates a 'dist' folder with index.html, favicon, and the js files/folder.  

Do you mean it automatically makes / updates the elegantWebpage.h in the src folder automatically?

closed 2020-09-10 by IAmOrion