ayushsharma82/AsyncElegantOTA #36: [HTTP ERROR] Bad Request at the end of updating

Zaborejszyn, 2020-09-24:
I'm getting "[HTTP ERROR] Bad Request" warning after percentage of load bar went to 100%. I can't locate the issue. I'm using example code on esp8266, bin file creates using PlatformIO
[image]

ayushsharma82, 2021-03-20:
This can be related to signal strength. AsyncElegantOTA has an inbuilt MD5 check which prevents the upload of corrupted firmware while it's being transferred.

yangminglong, 2021-03-26:
I had the same problem

augisbud, 2021-04-16:
how did you fix it? @yangminglong

Askhento, 2021-05-14:
Same here.
Exception 29. 
In the browser POST 400 (Bad Request). It does not seem to relate to signal strength. I am sitting right next to the router with -49 dBm signal.

Bjorn-J, 2021-06-18:
For me _Firmware_ update has this issue but not _Filesystem_. A few moths ago it worked in both cases, no library updates since then.
I'm using the procedure described at https://randomnerdtutorials.com/esp8266-nodemcu-ota-over-the-air-arduino/.

Ariasalf, 2021-10-21:
when I use Firefox, I'm getting "[HTTP ERROR] Bad Request" warning after the percentage of the load bar went to 100%. I'm using example code on esp8266, bin file created using Arduino IDE 1.8.15. (many retries the same behavior)

If I use chrome the upload is stoped and in the serial windows I can see the following error:

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Unhandled C++ exception: OOM

>>>stack>>>

ctx: sys
sp: 3fffebb0 end: 3fffffb0 offset: 0000
3fffebb0:  3fffb7c4 00000002 3fffec60 40100af0  
3fffebc0:  000000fe 00000000 00000000 00000000  
.
.
.
3fffffa0:  3fffdad0 00000000 3ffeee90 40209722  
<<<stack<<<

last failed alloc call: 40203799(80)

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

last failed alloc caller: 0x40203799

 ets Jan  8 2013,rst cause:2, boot mode:(3,6)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0005a3d0
~ld

ayushsharma82, 2021-12-18:
No Idea guys. Didn't face this issue with my deployed ESPs. You can try the latest version of the library and let me know if that also doesn't fix it.

Please reopen the issue if it's still not solved.

closed 2021-12-18 by ayushsharma82

lolwheel, 2021-12-23:
I've got the same issue - happens on Firefox. Chrome doesn't seem to be affected but the OTA still fails there.

mogoreanu, 2022-01-11:
Getting a similar error trying to update a custom firmware using chrome on a H801 board, the same procedure worked on a NodeMcu 0.9 and on a generic esp8266 board.

Serial log:
 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0008b5e0
~ld

vladimir-velinov, 2022-04-20:
I have the same 'Bad request' problem. I'm using esp8266-01 generic. After I upload the sketch through the USB/UART board then switch it on to an external power supply it works and I can access the update page. But when I try to upload the bin file then it fails with this error. Any consecutive attempt halts the progress bar at some percentage like 70-80% and then just hangs. Interestingly the esp itself does seem to operate as a refresh of the page serves it without problem.

Anyone reached some conclusion here? Btw is it a problem if in the loop() I have delay() as I've seen before for webservers usually the loop shouldn't be blocked for a longer period. Any suggestions?

AlphatestSylvain, 2022-05-04:
I had the same problem and I understood why.
Using PlatformIO, I introduiced `board_build.partitions = partitions.csv` in platformio.ini, to arrange a different partition memory.

Without this line, I've no more this issue.
Right now, I have to arrange the right partition file.

AlphatestSylvain, 2022-05-04:
Ok, I wrote a bad "partitions.csv" file.

In this blog, I read that :
"_It’s important to mention that the both OTA partitions should have the same size_ ..."
[https://blog.espressif.com/how-to-use-custom-partition-tables-on-esp32-69c0f3fa89c8](url)

And it was not the case fo my previous "partitions.csv" file...

keithw421, 2022-06-21:
I put esp in flash mode I.e. (iGPIO0 TO GND). It worked

trader-dave, 2022-11-10:
I installed this excellent lib last week into my app and it worked great. Yesterday I tried it on another ESP32 (also a devkit V1) and I am getting this error every time.  I then tried it on a third ESP32 - with the same .bin file - and it worked.  So in my little experiment, three identical ESP32's with the same .bin file, attempting to OTA flash from the same distance to my router - 2 boards work reliably, and 1 always fails with this error... I can't imagine what would be different with that one ESP32?

06GitHub, 2023-06-14:
Same as @techscapades 👍 for me : **error was due to incorrect partitions table**.

I was getting "[HTTP ERROR] Bad Request" error after percentage of load bar went to 100% when loading (16Mb) flash data (loading sketch firmware was OK).

After investigation, it appeared that my customized partition table was incorrect : it missed 4 bytes for spiffs size, 0xD0FFFC instead of 0xD10000

Once partitions table has been corrected, error "[HTTP ERROR] Bad Request" no longer occured when updating (16Mb) flash data.

**Incorrect partitions table**
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x170000,
app1,     app,  ota_1,   0x180000,0x170000,
spiffs,   data, spiffs,  0x2F0000,0xD0FFFC,
```

**Correct partitions table**
```
# Name,   Type, SubType, Offset,  Size, Flags
nvs,      data, nvs,     0x9000,  0x5000,
otadata,  data, ota,     0xe000,  0x2000,
app0,     app,  ota_0,   0x10000, 0x170000,
app1,     app,  ota_1,   0x180000,0x170000,
spiffs,   data, spiffs,  0x2F0000,0xD10000,
```

**Configuration**
ESP32-S3 module, 8Mb PSRAM, 16Mb Flash
Arduino IDE 1.8.19
arduino-esp32 2.0.8 https://github.com/espressif/arduino-esp32