ayushsharma82/AsyncElegantOTA #38: Reboots at 44%

mbobo, 2020-10-22:
Currently when I add this into my code and try and run it causes a panic and reboot.  Here is the offending line:

0x400da842: AsyncElegantOtaClass::begin(AsyncWebServer*, char const*, char const*)::{lambda(AsyncWebServerRequest*, String, unsigned int, unsigned char*, unsigned int, bool)#4}::operator()(AsyncWebServerRequest*, String, unsigned int, unsigned char*, unsigned int, bool) const at .pio/libdeps/espressif32/AsyncElegantOTA/src/AsyncElegantOTA.h line 11

I looked at that line number and looks like maybe it has an invalid data len?
I have a custom 2M partition set could that be my issue?

ayushsharma82, 2021-03-20:
@mbobo  That can be the cause of the problem. Are you leaving the exact same size for app1 (ota) partition as well?

ayushsharma82, 2021-12-18:
Closing due to inactivity.

closed 2021-12-18 by ayushsharma82