ayushsharma82/AsyncElegantOTA #39: Update failure with ESP32Cam

TRuHa83, 2020-11-05:
I've tested my sketch first and then your sample sketch, it always gives me 5% or even 0% error with an unexpected controller restart.

This is the error that appears in the terminal:

`IP address: 10.0.10.33
HTTP server started
abort() was called at PC 0x400e74fd on core 1

Backtrace: 0x4008efe0:0x3ffca850 0x4008f211:0x3ffca870 0x400e74fd:0x3ffca890 0x4008249f:0x3ffca8b0 0x400dfa9d:0x3ffca8d0 0x400df76c:0x3ffca8f0 0x400df9b7:0x3ffca910 0x400d221e:0x3ffca930 0x400d231d:0x3ffca980 0x401700a5:0x3ffca9c0 0x4016fdad:0x3ffcaa10 0x400d3bdb:0x3ffcaa40 0x400d5411:0x3ffcaaa0 0x400d55f9:0x3ffcaaf0 0x4014db85:0x3ffcab10 0x4014dbcd:0x3ffcab50 0x4014df2a:0x3ffcab70 0x4008af7d:0x3ffcaba0

Rebooting...`

I am using an ESP32Cam

ianiliff, 2020-11-18:
Try using a diffrent board. The partition scheme for the ESP32Cam does not allow OTA updates. See this thread right here:
https://esp32.com/viewtopic.php?t=14700

TRuHa83, 2020-12-20:
Thank you very much @ianiliff, I will look for an alternative to the code, it's a shame.

closed 2020-12-20 by TRuHa83

TRuHa83, 2021-02-19:
After searching and searching, in the end I have found the solution to make the microcontroller compatible. As @ianiliff says, the problem was in the partition, but not because of its own limitation, if not because of an incorrect configuration. Using VSCode and platformio.ini with the following settings it is possible to update it via OTA:

`platform = espressif32@1.9.0
board = esp32cam
framework = arduino
monitor_speed = 115200
board_build.f_cpu = 240000000L
board_build.partitions = min_spiffs.csv
build_flags = 
	-DBOARD_HAS_PSRAM
	-mfix-esp32-psram-cache-issue
lib_deps = 
	ayushsharma82/AsyncElegantOTA@^2.2.5`

eleminer, 2022-07-19:
is there also a method for the arduino ide?

TRuHa83, 2022-07-20:
> is there also a method for the arduino ide?
> 

Well, the truth is that I don't know a correct configuration for the Arduino IDE. I recommend you to use VSCode, in my opinion it is more complete and supports more languages, although it is true that the learning curve is higher. I'm sorry I can't help. 😔

eleminer, 2022-07-20:
yes, i agree with you. I have only one script which I have extended with OTA and that is only for Arduino IDE, with Plattformio it should cause problems. (according to readme)