ayushsharma82/AsyncElegantOTA #41: String version of the html rather than a byte/int array

ChiefyChief23, 2020-11-19:
Would it be possible to retrieve or at least be provided with the raw html that the Web server is using?

vshumilov, 2021-04-03:
Hi, ChiefyChief23

Overall, I liked this library. But when I saw a huge amount of obfuscated vue.js code put into PROGMEM, I was very disappointed. I don't understand the point of obfuscating js code at all.
It seems to me that the library is a "memory killer" ...
So I edited it and it freed about ~ 5% of the flash (and it's not a limit).

You could try to replace this lines in AsyncElegantOTA.h:
`                AsyncWebServerResponse *response = request->beginResponse_P(200, "text/html", _ELEGANT_HTML, ELEGANT_HTML_SIZE_); `
`               response->addHeader("Content-Encoding", "gzip");`
`    request->send(response);`

with this lines:
`String msg = "<!DOCTYPE html><html lang="en"><head>...</head>"; `
` msg += "<body> /* put your's html page with styles & js which 'POST' your's .bit file to server */ </body></html>";`
`AsyncWebServerResponse *response = request->beginResponse(200, "text/html", msg);`
                 `request->send(response);`

For read your's html from SPIFFS replace library original lines with:
`                AsyncWebServerResponse *response = request->beginResponse(SPIFFS, "/update.html", "text/html");`
`request->send(response);
`

!!! Don't forget delete or remark 
`#include "elegantWebpage.h"`

ayushsharma82, 2021-04-03:
@vshumilov , That's not obfuscation, it's compression. You should be glad that we were able to gzip a 1 mb webpage into a 40KB compressed byte array. ( and the source of webpage is provided in ui folder incase anybody wonders about it )

Having string based html for embedded microcontrollers is not the best design choice when you are including various styles and JavaScript along with the whole package. You will quickly reach a huge webpage size as you add more lines to it and blow past the SRAM capacity of esp32.

ayushsharma82, 2021-04-03:
@ChiefyChief23  Tip: Always check past issues about the topic before posting the question.

It's in the ui folder.

closed 2021-04-03 by ayushsharma82