ayushsharma82/AsyncElegantOTA #44: Not Rebooting, also cannot use setID

raako71, 2021-01-20:
I am getting a page saying success, although I can see my code hasn't updated. Possible to include a manual reboot button?

also, cannot use ElegantOTA.setID("abcd123"); .. or AsyncElegantOTA.setID("abc");
would be really useful to be able to enter a string to show the code version

frankjoshua, 2021-02-27:
I'm having the same issue with rebooting. I'm using a esp32 in a m5stickc.

ayushsharma82, 2021-03-20:
Hey @raako71 ,

Can you please check with a strong wifi connection as well? Sometimes the OTA fails if the signal strength is not good enough. 

Another possible cause can be where your code has registered `/update` async webserver route before the AsyncElegantOTA.

raako71, 2021-04-02:
good signal and no url conflict, device is just failing to reset. possibly due to known issue with ESP32 and resetting. I wonder if a button would help

ayushsharma82, 2021-04-03:
@raako71 Is there any blocking code in the loop?

raako71, 2021-04-04:
there are no delay or while functions if thats what you mean?

ayushsharma82, 2021-04-06:
Yup

nelsontechnologies, 2021-05-04:
Has this issue been fixed yet?  It is not restarting after successful upload on ESP32.

sui77, 2021-06-16:
Mine does not restart as well. I quickly made a reboot endpoint like this:

`  server.on("/reboot", HTTP_GET, [](AsyncWebServerRequest *request){
    AsyncWebServerResponse *response = request->beginResponse(301); 
    response->addHeader("Location","/");
    request->send(response);
    delay(100);
    ESP.restart();
  });  `

sui77, 2021-06-16:
Ok, got it. Don't get tricked by the 'That's all!' in the README.md, you have to also add this in your arduino main loop - as shown in the examples. Otherwise no restart is triggered.

    void loop(void) {
      AsyncElegantOTA.loop();
    }

ayushsharma82, 2021-06-21:
I think the current suicide method for ESP32 lead to this issue. I'll soon merge @sui77's pull request after testing it on my ESP32.

raako71, 2021-06-21:
best avoid the suicide method 😅

closed 2021-12-18 by ayushsharma82