ayushsharma82/AsyncElegantOTA #49: In about 50% of the cases I get "[HTTP ERROR] Bad Request"

TPD95100, 2021-04-03:
When using AsyncElegantOTO and ESP32 Dev Board, in about 50% of the cases I get "[HTTP ERROR] Bad Request" on the upload page after uploading the bin. 

I installed version v2.2.5

I found the same issue under https://github.com/ayushsharma82/AsyncElegantOTA/issues/31 - it appears that it was fixed in version v2.2.4 and may be, it is back now.

I tried to downgrade to version v2.2.4 in Arduino IDE. It was impossible to compile the code - it threw out tons of error messages that things were not declared (the type of endless messages one gets if you e.g. forgot to end a comment part of your code ... so, all in all, it is a fantastic tool if it works, but currently very unpredictable/unreliable.

ayushsharma82, 2021-06-19:
@TPD95100  AsyncElegantOTA uses WDT timer for restarting ESP32 in v2.2.5. #59 should fix your error I think.

Tejas-MD, 2021-06-25:
I'm having this prob a way too often. Please merge the PR and create a release asap.

ayushsharma82, 2021-06-25:
@Tejas-MD  Done. Refer to v2.2.6

ayushsharma82, 2021-06-25:
@TPD95100  A test with this version maybe? Let me know if you get some time to test this.

ayushsharma82, 2021-12-18:
Closing due to inactivity.

closed 2021-12-18 by ayushsharma82

Rabter1, 2022-03-25:
On version 2.2.6 i get the same error. 
Sometimes it works, sometimes the watchdog is triggered. 
I upload a rather big firmware.bin (1,84 MB)