ayushsharma82/AsyncElegantOTA #51: Strange behavior and question

mottibz, 2021-04-08:
Hello,

1. I am experiencing a strange behavior. I am trying to use this library on an ESP32 using AsyncWebServer.
When I select the file for update (either spiffs or firmware), it starts the OTA immediately and also immediately displays that 100% was done. Then, it takes some time (in which I don't know what is going on as there is no indication) until it shows the success page and then reboots the ESP32. Any suggestion how I can make it work as expected, meaning that the gauge will work as the OTA process executes and will reach 100% only when all is done. Thanks!!!

2. Any way to replace the ElegantOTA logo with something else?

Thanks a lot, Motti

ddv2005, 2021-05-26:
Hello,

I have the same issue - immediately displays 100% and then update in background.

mottibz, 2021-05-26:
Hi, No answer yet... Can we get some help? Thanks!

ddv2005, 2021-05-26:
Looks like I found the problem - antivirus (in my case Kasperskiy). It cache post request into memory and report 100%  immediately.

mottibz, 2021-05-26:
Interesting... I also use Kaspersky anti virus. I guess that either the AsyncElegantOTA guys overcome this issue or we will need to find another OTA approach...

ayushsharma82, 2021-06-21:
I'll try implementing the 'chunks' approach in a future release. I hope it will be simple.

closed 2021-12-18 by ayushsharma82