ayushsharma82/AsyncElegantOTA #58: Upload and Always Blue Light On After This Library

haliltaskit, 2021-06-13:
Hello all ,

After I added this library  - my Nodemcu v3 always showed blue light on and didn't allow me to upload new code. It looks like it stucked on loop and I can't do anything on this . Is there someone who have any idea or experience about this ?

ayushsharma82, 2021-06-13:
@haliltaskit  This only happens when you have not rebooted your nodemcu after flashing from PC. Make sure to reboot it once before performing OTA.

haliltaskit, 2021-06-13:
Thank you for response @ayushsharma82 . 

I can't access to my Nodemcu now from my Mac after AsyncElegantOTA implementation. I see my device on ports after plugged but can't connect.
I got this error. (   File "/Users/hitaskit/Library/Arduino15/packages/esp8266/hardware/esp8266/2.7.4/tools/esptool/esptool.py", line 529, in connect
    raise FatalError('Failed to connect to %s: %s' % (self.CHIP_NAME, last_error))
esptool.FatalError: Failed to connect to ESP8266: Timed out waiting for packet header )

[image]

What shall I try , any idea ?

ayushsharma82, 2021-06-13:
@haliltaskit 
One fact about ESP8266 is that we can't brick it with any malfunctioning code, If it's not able to go into firmware download mode via serial, then it's probably a hardware issue.

If there are any jumpers or peripherals connected to it, try removing them one by one.

haliltaskit, 2021-06-13:
I don't have any connected wires , sadly I can't access OTA or serial anymore.

ayushsharma82, 2021-06-14:
Please make sure you have the following things correct:
1. Make sure you have selected Nodemcu as the board in Arduino IDE.
2. Press down the 'Boot' button while plugging it in your usb port.