ayushsharma82/AsyncElegantOTA #63: Reboots at 39%

mozgy, 2021-07-22:
Flashing firmware.bin crashes/reboots with 39% shown on web page, flashing SPIFFS works fine tho,

```
abort() was called at PC 0x401740d9 on core 1

Backtrace: 0x40090e18:0x3ffee8a0 0x40091091:0x3ffee8c0 0x401740d9:0x3ffee8e0 0x4008c2ce:0x3ffee900 0x40173f7a:0x3ffee920 0x400eae71:0x3ffee940 0x400e5804:0x3ffee960 0x400e5a05:0x3ffee980 0x400d2af5:0x3ffee9a0 0x400d2c2d:0x3ffee9f0 0x401a0789:0x3ffeea40 0x401a0445:0x3ffeea90 0x400d9059:0x3ffeeac0 0x400dab85:0x3ffeeb30 0x400dad5d:0x3ffeeb80 0x40195d1a:0x3ffeeba0 0x40195d99:0x3ffeebe0 0x401964b2:0x3ffeec00 0x40092d22:0x3ffeec30

Decoding stack results
0x40090e18: invoke_abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 156
0x40091091: abort at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/esp32/panic.c line 171
0x401740d9: std::basic_istream   >::~basic_istream() at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/istream line 104
0x4008c2ce: spi_flash_erase_range at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/spi_flash/flash_ops.c line 220
0x40173f7a: std::__convert_to_v (char const*, double&, std::_Ios_Iostate&, int* const&) at c++locale.cc line 149
0x400eae71: fde_mixed_encoding_compare at /builds/idf/crosstool-NG/.build/src/gcc-5.2.0/libgcc/unwind-dw2-fde.c line 362
0x400e5804: WebServer::_parseFormUploadAborted() at C:\Users\xx\.platformio\packages\framework-arduinoespressif32\libraries\WebServer\src\Parsing.cpp line 587
0x400e5a05: WebServer::_parseForm(WiFiClient&, String, unsigned int) at C:\Users\xx\.platformio\packages\framework-arduinoespressif32\libraries\WebServer\src\Parsing.cpp line 379
0x400d2af5: AsyncElegantOtaClass::begin(AsyncWebServer*, char const*, char const*) at c:\users\xx\.platformio\packages\toolchain-xtensa32@2.50200.97\xtensa-esp32-elf\include\c++\5.2.0/functional line 2259
0x400d2c2d: AsyncElegantOtaClass::AsyncElegantOtaClass() at .pio\libdeps\esp32cam\AsyncElegantOTA\src/AsyncElegantOTA.h line 29
0x401a0789: std::time_get    > >::date_order() const at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/locale_facets_nonio.h line 405
0x401a0445: std::moneypunct ::negative_sign() const at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/locale_facets_nonio.h line 1183
0x400d9059: AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*) at .pio\libdeps\esp32cam\ESP Async WebServer\src\WebRequest.cpp line 70
0x400dab85: AsyncAbstractResponse::_ack(AsyncWebServerRequest*, unsigned int, unsigned int) at .pio\libdeps\esp32cam\ESP Async WebServer\src\WebResponses.cpp line 295
0x400dad5d: AsyncFileResponse::~AsyncFileResponse() at .pio\libdeps\esp32cam\ESP Async WebServer\src\WebResponses.cpp line 479
0x40195d1a: sdmmc_send_cmd_send_cid at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/sdmmc/sdmmc_cmd.c line 209
0x40195d99: sdmmc_send_cmd_set_relative_addr at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/sdmmc/sdmmc_cmd.c line 230
0x401964b2: sdmmc_send_cmd_switch_func at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/sdmmc/sdmmc_sd.c line 139
0x40092d22: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143

```
crash decoded using arduino as pio monitor one does not work ..

ayushsharma82, 2021-08-04:
Are you using a different partition table?

mozgy, 2021-08-05:
I'm using AI Thinker Cam, there's no available choice there.

ayushsharma82, 2021-08-28:
Please refer to #39

closed 2021-08-28 by ayushsharma82