ayushsharma82/AsyncElegantOTA #65: Content Encoding Error

pbliampas, 2021-07-29:
After I connect to the local IP of my ESP32 device and go to: '"esp32localIP"/update' page,
the update page doesn't display anything most of the times, and an error appears on the browser canvas:

"Content Encoding Error

The page you are trying to view cannot be shown because it uses an invalid or unsupported form of compression.

-Please contact the website owners to inform them of this problem."

Sometimes, it does not even display this error, it's just blank page, and starts automatically downloading a file called "update".

I have tried Google Chrome, Mozilla Firefox and Microsoft Edge.

Any other with the same issue?

pbliampas, 2021-07-29:
For some reason, through the developer tools, it seems to recognize the "/update" as a document. Is this any helpful?

[image]

pbliampas, 2021-07-29:
And sometimes it's not loading at all, showing this error: "ERR_CONTENT_DECODING_FAILED"
[image]

ayushsharma82, 2021-08-04:
@pbliampas  Please make sure there is no `delay` or `loop` function which takes too long to execute in your code.

ayushsharma82, 2021-12-18:
Closing due to inactivity.