ayushsharma82/AsyncElegantOTA #66: Filesystem update

KLelong, 2021-08-09:
Hi,
I would like to use your library, but I have a question : 
how does the update of the file-system works ? What I mean is : are the files in the update-file replaced, or is the whole file-system wiped and then populated with the new files ?
I ask because I'm using the file-system to store configurations. Those configurations are not in the update-file. So, when the file-system is wiped, I lose those. If the files are only replaced, I could use the library.
E.g.
file-system : 
file1, file2, config
update-file.bin : 
file1, file2
what is in the file-system after the update-file.bin is applied ?
TIA

ayushsharma82, 2021-08-09:
Hi,

The whole file system gets overwritten with SPIFFS/TinyFS

Regards,
Ayush

KLelong, 2021-08-09:
Thanks,
I'll have to work around that then.
Is there a way to hide the "filesystem" radio-button ?