ayushsharma82/AsyncElegantOTA #68: UI Update

nelsontechnologies, 2021-08-19:
After build for production in UI, what utility do you use to convert to hex to paste into elegantwebpage.h? In the dist folder, i have favicon, index.html and js. How do i merge these to the hex? THank you in advance!

06GitHub, 2021-11-06:
**elegantWebpage.h PROGMEM ELEGANT_HTML** variable represents gzip file exchanged between Web server and Web client :
```
const uint8_t ELEGANT_HTML[] PROGMEM = { 
31,139,8,0,0,0,0,0,2,3,148,58,233,122,26,185,178,255,243,20,141,206,249,56,221,99,185,1,219,113,
[snip]
62,203,238,255,227,255,1,204,222,188,142,16,149,2,0 
};
```

The first two bytes are "31" and "139", so in hex "1F" and "8B", which are the headers of gzip file https://www.rfc-editor.org/rfc/rfc1952 :

>  2.3.1. Member header and trailer
>          ID1 (IDentification 1)
>          ID2 (IDentification 2)
>             These have the fixed values ID1 = 31 (0x1f, \037), ID2 = 139
>             (0x8b, \213), to identify the file as being in gzip format.

By dumping in a binary file ELEGANT_HTML data,  one creates a gz file (ie zipped file) which can be uncompressed for instance with 7-Zip on Windows 10 https://www.7-zip.org/

Once uncompressed, html page is visible in text format :

```
<!DOCTYPE html>
<html lang=en>

<head>
    <meta charset=utf-8>
    <meta http-equiv=X-UA-Compatible content="IE=edge">
    <meta name=viewport content="width=device-width,initial-scale=1">
    <link rel=icon href=/favicon.ico> <title>ElegantOTA</title>
    <script data-name="BMC-Widget" async src="https://cdnjs.buymeacoffee.com/1.0.0/widget.prod.min.js" data-id="6QGVpSj" data-description="Support me on Buy me a coffee!" data-message="You can always support my work by buying me a coffee!" data-color="#FF813F" data-position="right" data-x_margin="24" data-y_margin="24"></script>
</head>

<body style="overflow: hidden;">
    <noscript>
        <strong>We're sorry but ElegantOTA doesn't work properly without JavaScript enabled. Please enable it to continue.</strong>
    </noscript>
    <div id=app></div>
    <script defer>(function(t){var e={};function r(n){if(e[n])return e[n].exports;var o=e[n]={i:n,l:!1,exports:{}};return t[n].call[snip]
/*!
 * Vue.js v2.6.11
 * (c) 2014-2019 Evan You
 * Released under the MIT License.
 */
var r=Object.freeze({});function n(t){return void 0===t||null===t}function o(t){return void 0!==t&&null!==t}function i(t) [snip]</script>
</body>
</html>
```

alesha76, 2021-11-15:
How to create a binary file, tell me.

06GitHub, 2021-11-16:
I used Python to process PROGMEM ELEGANT_HTML data into binary file, see code below. 
Code can be done with any Python IDE, I used Anaconda https://www.anaconda.com/ with Jupyter notebook https://jupyter.org/
Jupyter notebook is part of Anaconda installation

```
"""
Write PROGMEM ascii data in gzip binary file with map
INPUT
=====
eota.h
31,139,8,0,0,0,0,0,2,3,148,58,233,122,26,185,178,255,243,20,141,206,249,56,221,99,185,1,219,113,
98,136,236,155,197,158,53,147,57,217,102,33,190,254,68,119,1,74,26,137,145,4,14,99,243,238,183,164,110,209,
128,177,103,238,44,109,45,165,82,237,85,146,120,244,172,241,234,205,203,247,191,255,114,30,141,237,164,56,125,244,
...

OUTPUT
======
eota.gz

"""

with open('data/eota.gz', 'w+b') as eotazip:
    with open('data/eotad.h', "r") as eota_h:
        for line in eota_h:
            #print (line)
            data = list(map(int,line.split(",")))
            print (data)
            binary_format = bytearray(data)
            print (binary_format)
            eotazip.write(binary_format)
    eota_h.close()
eotazip.close()
```

"print (data)" results
`[31, 139, 8, 0, 0, 0, 0, 0, 2, 3, 148, 58, 233, 122, 26, 185, 178, 255, 243, 20, 141, 206, 249, 56, 221, 99, 185, 1, 219, 113, 98, 136, 236, 155, 197, 158, 53, 147, 57, 217, 102, 33, 190, 254, 68, 119, 1, 74, 26, 137, 145, 4, 14, 99, 243, 238, 183, 164, 110, 209, 128, 177, 103, 238, 44, 109, 45, 165, 82, 237, 85, 146, 120, 244, 172, 241, 234, 205, 203, 247, 191, 255, 114, 30, 141, 237, 164, 56, 125, 244, ...`

"print (binary_format)" results
`bytearray(b'\x1f\x8b\x08\x00\x00\x00\x00\x00\x02\x03\x94...`

ayushsharma82, 2021-12-18:
UI of webpage is available in the parent repo: https://github.com/ayushsharma82/ElegantOTA/tree/master/ui

It's a vuejs SPA. You can compile it by running the following commands in that directory:
```
npm install
npm run build
```

Once compiled, it will automatically update the `elegant_webapge.h`.

closed 2021-12-18 by ayushsharma82