ayushsharma82/AsyncElegantOTA #72: [BUG] :  OTA upload starts but never goes to the end.

HWO, 2021-09-14:
[//]: <> (Please follow this template for fast resolutions)
**Describe the bug**
The OTA upload never goes through and gets stuck between 50 ~ 70% progress.
I have the same phenomenon on several ESP01 cards.
The sample code provided by the lib does the same.

**Do you get any error/exception stack? If so please paste it here:**
No upload error in the browser console.

</details>

**Microcontroller Specs:**
 - Board: ESP01
 - Microcontroller: ESP8266
 - Did you modify the partition table? No
 - Browser: [chrome/edge]
 - Library Version: latest

**Additional context**
On the page `http://xxx.xxx.x.xx/update` after having selected the .bin file, the upload starts but never goes to the end.

HWO, 2021-09-15:
> **Describe the bug**
> The OTA upload never goes through and gets stuck between 50 ~ 70% progress.
> I have the same phenomenon on several ESP01 cards.
> The sample code provided by the lib does the same.
> 
> **Do you get any error/exception stack? If so please paste it here:**
> No upload error in the browser console.
> 
> **Microcontroller Specs:**
> 
> * Board: ESP01
> * Microcontroller: ESP8266
> * Did you modify the partition table? No
> * Browser: [chrome/edge]
> * Library Version: latest
> 
> **Additional context**
> On the page `http://xxx.xxx.x.xx/update` after having selected the .bin file, the upload starts but never goes to the end.

giuliano2505, 2021-11-24:
Adding info to this bug, I try updating an ESP-01s with 1MB flash, my sketch uses 566KB and my ldscript is "eagle.flash.1m256.ld". So I have to asume that is not posible to update if sketch is bigger that fs partition? Or OTA partition?

ayushsharma82, 2021-12-18:
@giuliano2505 Correct.

ayushsharma82, 2021-12-18:
@HWO Can you check if you have an OTA partition for your ESP01 board? This issue is mostly due to low flash memory.

tahunus, 2022-01-07:
**Same issue**
Board: DOIT ESP32 DEVKIT V1
Partition Table: not an option to change in Arduino IDE
Browser: Edge
AsyncElegantOTA ver: 2.2.6

From compiler: 
```
Sketch uses 1084725 bytes (82%) of program storage space. Maximum is 1310720 bytes.
Global variables use 42252 bytes (12%) of dynamic memory, leaving 285428 bytes for local variables. Maximum is 327680 bytes.
```
If I understood above comments, with 82% of memory used by the .bin file, there's no space for OTA? Or should I get another ESP32 that allows for memory partition?

Thanks!

darasolution, 2022-02-14:
Same issue using a Wemos D1 ESP8266 . "normal" OTA works but i can't use Webserial. Only using ~ 40% of flash so this couldn't be the problem?

watersoup, 2023-01-23:
I am having same issue with OTA notable to finish even for filesystem which are <5kb. Where as the size of the firmware on ESP01-1m ESP8266ex loading stepper functionality.
RAM:   [=====     ]  46.2% (used 37816 bytes from 81920 bytes)
Flash: [=======   ]  68.2% (used 519224 bytes from 761840 bytes)

DONGILBAE, 2023-06-23:
I got a problem and solved. 
It was my mistake. 
It is all fine and run and update well. 
My mistake was compile condition On arduino ide. 
I changed my condition as Hug APP to 16M Flash(3M APP/9.9M FATFS ) 
  well well my esp32 has 16M flash memory 
So body want to change condition , you have to check your esp32 memory ( have 4M, and 8M, 16M )
any way, depend on me, my esp32 update well and run well. 

if your esp32 memory just 4M and  your program big than 1.2M, you can not update well. 
So, you have to change memory upper size than 4M. 
if your program size smaller than 1.2M... it is ok... you can run well. 

any way thanks for this iussue