ayushsharma82/AsyncElegantOTA #75: error compiling for nodemvu and di mini

donbrew1, 2021-11-04:


donbrew1, 2021-11-04:
I tried on a D1 mini and a nodemcu esp8266 and got the error, the sketches I tried did compile without the include statement (cut and pasted from your instructions.

error msg:
Arduino: 1.8.13 (Windows 10), Board: "LOLIN(WEMOS) D1 R2 & mini, 80 MHz, Flash, Disabled, All SSL ciphers (most compatible), 4M (1M SPIFFS), v2 Lower Memory, Disabled, None, Only Sketch, 921600"

C:\Program Files\Arduino\arduino-builder -dump-prefs -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Users\donwin7\AppData\Local\Arduino15\packages -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\donwin7\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\donwin7\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M1M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\donwin7\AppData\Local\Temp\arduino_build_383935 -warnings=none -build-cache C:\Users\donwin7\AppData\Local\Temp\arduino_cache_834288 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.python.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -verbose C:\Users\donwin7\AppData\Local\Temp\arduino_modified_sketch_464432\sketch_nov04a.ino

C:\Program Files\Arduino\arduino-builder -compile -logger=machine -hardware C:\Program Files\Arduino\hardware -hardware C:\Users\donwin7\AppData\Local\Arduino15\packages -tools C:\Program Files\Arduino\tools-builder -tools C:\Program Files\Arduino\hardware\tools\avr -tools C:\Users\donwin7\AppData\Local\Arduino15\packages -built-in-libraries C:\Program Files\Arduino\libraries -libraries C:\Users\donwin7\Documents\Arduino\libraries -fqbn=esp8266:esp8266:d1_mini:xtal=80,vt=flash,exception=disabled,ssl=all,eesz=4M1M,ip=lm2f,dbg=Disabled,lvl=None____,wipe=none,baud=921600 -vid-pid=1A86_7523 -ide-version=10813 -build-path C:\Users\donwin7\AppData\Local\Temp\arduino_build_383935 -warnings=none -build-cache C:\Users\donwin7\AppData\Local\Temp\arduino_cache_834288 -prefs=build.warn_data_percentage=75 -prefs=runtime.tools.xtensa-lx106-elf-gcc.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.xtensa-lx106-elf-gcc-2.5.0-3-20ed2b9.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\xtensa-lx106-elf-gcc\2.5.0-3-20ed2b9 -prefs=runtime.tools.python.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.python-3.7.2-post1.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\python\3.7.2-post1 -prefs=runtime.tools.mkspiffs.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -prefs=runtime.tools.mkspiffs-2.5.0-3-20ed2b9.path=C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\tools\mkspiffs\2.5.0-3-20ed2b9 -verbose C:\Users\donwin7\AppData\Local\Temp\arduino_modified_sketch_464432\sketch_nov04a.ino

Using board 'd1_mini' from platform in folder: C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2

Using core 'esp8266' from platform in folder: C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2

Detecting libraries used...

"C:\\Users\\donwin7\\AppData\\Local\\Arduino15\\packages\\esp8266\\tools\\xtensa-lx106-elf-gcc\\2.5.0-3-20ed2b9/bin/xtensa-lx106-elf-g++" -D__ets__ -DICACHE_FLASH -U__STRICT_ANSI__ "-IC:\\Users\\donwin7\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/include" "-IC:\\Users\\donwin7\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/lwip2/include" "-IC:\\Users\\donwin7\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2/tools/sdk/libc/xtensa-lx106-elf/include" "-IC:\\Users\\donwin7\\AppData\\Local\\Temp\\arduino_build_383935/core" -c -w -Os -g -mlongcalls -mtext-section-literals -fno-rtti -falign-functions=4 -std=c++11 -ffunction-sections -fdata-sections -fno-exceptions -w -x c++ -E -CC -DNONOSDK221=1 -DF_CPU=80000000L -DLWIP_OPEN_SRC -DTCP_MSS=536 -DLWIP_FEATURES=1 -DLWIP_IPV6=0 -DARDUINO=10813 -DARDUINO_ESP8266_WEMOS_D1MINI -DARDUINO_ARCH_ESP8266 "-DARDUINO_BOARD=\"ESP8266_WEMOS_D1MINI\"" -DFLASHMODE_DIO -DESP8266 "-IC:\\Users\\donwin7\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\cores\\esp8266" "-IC:\\Users\\donwin7\\AppData\\Local\\Arduino15\\packages\\esp8266\\hardware\\esp8266\\2.5.2\\variants\\d1_mini" "C:\\Users\\donwin7\\AppData\\Local\\Temp\\arduino_build_383935\\sketch\\sketch_nov04a.ino.cpp" -o nul -DARDUINO_LIB_DISCOVERY_PHASE

[... cut ...]
Using library ESPAsyncTCP-master at version 1.2.2 in folder: C:\Users\donwin7\Documents\Arduino\libraries\ESPAsyncTCP-master 

Using library ESPAsyncWebServer-master at version 1.2.3 in folder: C:\Users\donwin7\Documents\Arduino\libraries\ESPAsyncWebServer-master 

Using library Hash at version 1.0 in folder: C:\Users\donwin7\AppData\Local\Arduino15\packages\esp8266\hardware\esp8266\2.5.2\libraries\Hash 

Using library AsyncElegantOTA at version 2.2.6 in folder: C:\Users\donwin7\Documents\Arduino\libraries\AsyncElegantOTA 

exit status 1

Error compiling for board LOLIN(WEMOS) D1 R2 & mini.

dineshvyas, 2021-12-04:
I am facing same issue for esp12e.

ayushsharma82, 2021-12-18:
It's related to https://github.com/esp8266/Arduino/issues/6939.

Please update your ESP8266 arduino core in boards manager.

ayushsharma82, 2021-12-18:
Tested with ESP8266 Core v3.0.2 and works fine. Hence closing this issue.

Please reopen it if still not fixed.

closed 2021-12-18 by ayushsharma82