ayushsharma82/AsyncElegantOTA #78: [BUG] WiFiManger and AsyncElegantOTA do not play well together.

markpope, 2021-12-20:
I combined two working examples, WiFiManager and AsyncElegantOTA, producing one app that panic’s the core. Is there a combability issue or my coding. Two great features should play well together. Others have verified the same problem.

I get two stack dump:

Decoding stack results
0x4022f20f: __strftime at /workdir/repo/newlib/newlib/libc/time/../time/strftime.c line 715
0x40226b4d: _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 204

And

Decoding stack results
0x4025ad31: tcp_input at core/tcp_in.c line 702
0x4025ad31: tcp_input at core/tcp_in.c line 702
0x402242d9: uw_update_context_1 at /workdir/repo/gcc-gnu/libgcc/unwind-dw2.c line 208
0x4022f20f: __strftime at /workdir/repo/newlib/newlib/libc/time/../time/strftime.c line 715
0x40226b4d: _printf_i at /workdir/repo/newlib/newlib/libc/stdio/nano-vfprintf_i.c line 204
0x402242d9: uw_update_context_1 at /workdir/repo/gcc-gnu/libgcc/unwind-dw2.c line 208
0x4024a935: std::num_get    > >::_M_extract_int (std::istreambuf_iterator   >, std::istreambuf_iterator   >, std::ios_base&, std::_Ios_Iostate&, long&) const at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 4186
0x4024a947: std::num_get    > >::_M_extract_int (std::istreambuf_iterator   >, std::istreambuf_iterator   >, std::ios_base&, std::_Ios_Iostate&, long&) const at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/basic_string.h line 4186
0x40241bf3: std::time_get    > >::_M_extract_via_format(std::istreambuf_iterator   >, std::istreambuf_iterator   >, std::ios_base&, std::_Ios_Iostate&, tm*, char const*) const at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/ios_base.h line 182
0x40246157: std::collate ::~collate() at /workdir/arena.x86_64/gcc-gnu/xtensa-lx106-elf/libstdc++-v3/include/bits/locale_classes.h line 748

### Basic Infos

#### Hardware  
WiFimanager Branch/Release:  Master V0.16(Latest)

Esp8266/Esp32: 

Hardware:			NodeMCU 1.0 (ESP-12E)

Core Version:      	?

### Description

Panic core when trying to upload a new .bin image. Works fine when AsyncElegantOTA is not included.

### Settings in IDE

Module:  			NodeMcu

Additional libraries: 
         bbx10/DNSServer@^1.1.0
	ottowinter/ESPAsyncWebServer-esphome @ ^2.1.0
	tzapu/WiFiManager@^0.16.0
	ayushsharma82/AsyncElegantOTA@^2.2.6

### Sketch

```cpp
#BEGIN
#include <Arduino.h>
#include <ESP8266WiFi.h>
#include <ESP8266WebServer.h>
#define WEBSERVER_H
#include <ESPAsyncTCP.h>
#include <WiFiManager.h>
#include <ESPAsyncWebServer.h>
#include <AsyncElegantOTA.h>

AsyncWebServer server(80);

void setup(void) {
  Serial.begin(9600);

  WiFiManager wifiManager;
  wifiManager.autoConnect("CheesePress");

  Serial.println("");
  Serial.print("Connected to ");
  Serial.println(ssid);
  Serial.print("IP address: ");
  Serial.println(WiFi.localIP());

  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
    request->send(200, "text/plain", "Hi! I am ESP8266.");
  });

  AsyncElegantOTA.begin(&server);    // Start ElegantOTA
  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
  AsyncElegantOTA.loop();
}

#END
```

### Debug Messages

```
messages here
```

ayushsharma82, 2021-12-21:
Hi @markpope ,

That's initially because the sketch is trying to use 2 different type of web server libraries which are not meant to work together.

If your app is WebServer based ( WiFiManager uses WebServer ) then using non-async version of ElegantOTA would be the right choice.

AsyncElegantOTA is meant to be used for projects that rely heavily on AsyncWebServer.

I hope this was helpful.

markpope, 2021-12-21:
Thanks for the prompt response. My project does use AsyncWebServer and Azure IoT MQTT client. Is there anything I can do to include WiFiManager?

ayushsharma82, 2021-12-21:
These are alternatives that use AsyncWebServer:
1. https://github.com/alanswx/ESPAsyncWiFiManager
2. if you are aiming for simply connecting to WiFi then ESPConnect can be used: https://github.com/ayushsharma82/ESPConnect

markpope, 2021-12-25:
Ayush,

 I started reducing libraries to see if an incompatibility was causing the panic. I created a new PlatformIO project, including AsyncElegantOTA, me-no-dev/ESPAsyncWebServer, me-no-dev/ESPAsyncTCP, and the  example in the RandomNerdTutorials tutorial and get the same panic when the .bin completes uploading. I flashed and cleaned all libraries. Does OTA with the current libraries work for you?