ayushsharma82/AsyncElegantOTA #80: Update goes to 100%, but never goes to completion page

svenyonson, 2021-12-25:
Mac M1, using Firefox or Chrome. NodeMCU. 

If I turn on the dev console in chrome, I can see that the POST fails with a ERR_CONNECTION_RESET. 

However, the update itself worked. If I then go to the root page (/), I can see the change in the title that I made.

If I use Safari, the progress immediately goes to 100%. If I wait long enough, I can see that it also finishes the upload, but with "The network connection was lost"

ayushsharma82, 2022-01-21:
I think you are right, it's an issue with ElegantOTA which I'll tackle this with next update. The MCU restarts before actually sending the response back.

darasolution, 2022-02-14:
Same issue using a Wemos D1 ESP8266 . "normal" OTA works but i can't use Webserial then :/

jvl88, 2022-05-24:
I came across this issue as well.. Except in my case when I open the bin file it goes straight to 100% afterwards the ESP is unreachable and is corrupted.  Even happens with verified code which uploads and works fine via FTDI programmer. Any idea how I can solve this?