ayushsharma82/AsyncElegantOTA #83: source of the webpage

musapinar, 2022-01-19:
Hi,

Can you publish the sources for https://github.com/ayushsharma82/AsyncElegantOTA/blob/master/src/elegantWebpage.h ?
As far as i can see it has some packed VueJS code and also requests 3rd party assets (minified JS on a cloud).

Thanks

ayushsharma82, 2022-01-19:
Hi @musapinar ,

The source can be found in the parent repository: https://github.com/ayushsharma82/ElegantOTA/tree/master/ui

All the assets related to ElegantOTA are loaded from MCU, the rest you see in the network logger is `(BMC) Buy Me a Coffee` SDK loading contributions. Nothing to worry about, I'm cautious of these things as well :)

musapinar, 2022-01-19:
Thank you.
You might consider using SRI for that external file (https://developer.mozilla.org/en-US/docs/Web/Security/Subresource_Integrity)

I just wanted to take a look at how you handle the file upload progress bar, There seem to be nothing different than my code (xhr request and event handler onprogress). I log the progress percent both on client side via ```console.log()``` and on server side via ```Serial.print()``` and they do not seem accurate. Have you noticed weird behaviors while implementing that feature? 
On my end, I can see the browser console log go from 0% to 100% (had to throttle the connection to "regular 3g" in the network tab of the devtool), but the progress bar on the server side starts at 0% only after the browser side has reached 100%. I would have expected to be both synchronized but it's not. I'm not proficient enough in C to understand how it works under the hood but it's like it buffers the whole stream before writing it to the file system. Is it how it is supposed to work?

https://i.imgur.com/uwjozCA.gif

ayushsharma82, 2022-01-21:
It's already mentioned in the ReadMe under "Antivirus Issue".

ayushsharma82, 2022-01-21:
Regarding SRI, BMC keeps updating their js package so the hash will change every time. It's best to keep it without SRI for this case.

musapinar, 2022-01-21:
> It's already mentioned in the ReadMe under "Antivirus Issue".

I saw that and I'm not sure this is related. See my screencapture above. Can you confirm you experience the same exact behavior when simulating a low speed internet?
I've tried with 3 different devices, without any firewall/AV.

musapinar, 2022-01-21:
nevermind, it was my reverse proxy buffering requests.
thanks for sharing.

closed 2022-01-21 by musapinar