ayushsharma82/AsyncElegantOTA #87: [BUG]  ESP8266 Stuck.....

PhantomRex123, 2022-03-12:
[//]: <> (Please follow this template for fast resolutions)
**Describe the bug**
I'm running an ESP8266 and when i try and upload the code on my esp01 its stuck or either gives an error 

**Do you get any error/exception stack? If so please paste it here:**
<details>
  <summary>Exception</summary>

  [//]: <> (Paste exception under this line)
  
last failed alloc call: 4020B159(80)

--------------- CUT HERE FOR EXCEPTION DECODER ---------------

last failed alloc caller: 0x4020b159

 ets Jan  8 2013,rst cause:2, boot mode:(3,7)

load 0x4010f000, len 3460, room 16 
tail 4
chksum 0xcc
load 0x3fff20b8, len 40, room 4 
tail 4
chksum 0xc9
csum 0xc9
v0009a570

</details>

**Microcontroller Specs:**
 - Board: ESP01
 - Microcontroller:  8266
 - Did you modify the partition table? No
 - Browser: chrome
 - Library Version: latest on platformio which is .6

**Additional context**
Add any other context about the problem here.

PhantomRex123, 2022-03-12:
This has been killing me! the experienec isn't elegant so-far

amefs, 2022-03-17:
Have you tried [v2.2.5](https://github.com/ayushsharma82/AsyncElegantOTA/tree/v2.2.5)? It seems to work fine on my ESP8266.