ayushsharma82/AsyncElegantOTA #89: Converting to Ethernet

pscott-ams, 2022-03-22:
I am trying to modify the ESP32_Async_Demo example to work over ethernet instead of WiFi.

Hardware I am using:
- WeMos D1 Mini32
- W5500 Ethernet Module

Wired connections from ESP32 to W5500:
- GPIO5 -> CS
- GPIO18 -> SCK
- GPIO23 -> MO
- GPIO19 -> MI
- GPIO26 -> RST
- GPIO22 -> INT

Here is my code:

```
#include <Arduino.h>
#include <SPI.h>
#include <Ethernet.h>
#include <AsyncTCP.h>
#include <ESPAsyncWebServer.h>
#include <AsyncElegantOTA.h>

byte mac[] = { 0x02, 0xF0, 0x0D, 0xBE, 0xEF, 0x01 };
IPAddress ip = { 192, 168, 167, 101 };
EthernetUDP Udp;
AsyncWebServer server(80);

void setup(void) {
  Serial.begin(115200);
  
  SPI.begin();
  Ethernet.init(5);
  // start the Ethernet connection:
  Serial.println("Initializing Ethernet...");
  Ethernet.begin(mac, ip);
  Serial.print("Local IP: ");
  Serial.println(Ethernet.localIP());
  Serial.print("DNS: ");
  Serial.println(Ethernet.dnsServerIP());
  Serial.print("Gateway: ");
  Serial.println(Ethernet.gatewayIP());
  Serial.print("Subnet Mask: ");
  Serial.println(Ethernet.subnetMask());

  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
    request->send(200, "text/plain", "Hi! I am ESP32.");
  });

  AsyncElegantOTA.begin(&server);    // Start ElegantOTA
  server.begin();
  Serial.println("HTTP server started");
}

void loop(void) {
}
```

After uploading, the ESP32 crashes when `server.begin()` is called.
This is the output from the serial monitor:

```
Initializing Ethernet...
Local IP: 192.168.167.101
DNS: 192.168.167.1
Gateway: 192.168.167.1
Subnet Mask: 255.255.255.0
assertion "Invalid mbox" failed: file "/home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/lwip/lwip/src/api/tcpip.c", line 416, function: tcpip_api_call
abort() was called at PC 0x400e45bf on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x400852a4:0x3ffb1e50 0x40085521:0x3ffb1e70 0x400e45bf:0x3ffb1e90 0x4011f1b3:0x3ffb1ec0 0x4011548b:0x3ffb1ef0 0x400d696e:0x3ffb1f40 0x400d2283:0x3ffb1f60 0x400e223a:0x3ffb1fb0 0x40086532:0x3ffb1fd0

Rebooting...
```

simogaspa84, 2022-09-23:
Hi @pscott-ams ...

Were you able to fix this problem? 

I should do the same by using this lib 

https://github.com/JAndrassy/EthernetENC

sdrshnptl, 2022-12-19:
Hi @pscott-ams ,
Have you able to fix issue?
if yes please share.