ayushsharma82/AsyncElegantOTA #91: [BUG]

puhycz, 2022-04-09:
[//]: <> (Please follow this template for fast resolutions)
**Describe the bug**
I test with ESP8285, but it not works. Update over AsyncElegantOTA stop on 27%. Default example sketch. Is possible configure to use with ESP8285 (f.e. SonOff boards etc...)

**Microcontroller Specs:**
 - Board: [ESP8285 ]
 - Microcontroller:  [ ESP8285 ]
 - Did you modify the partition table? [No]
 - Browser: [chrome]
 - Library Version: [latest]

puhycz, 2022-04-09:
And with slow sketketche it go to 100%, but still error:
[image]

ayushsharma82, 2022-04-26:
Hi @puhycz ,
Have you selected the right partition table with OTA support in Arduino IDE? 

Important step is to flash it with Serial first and then proceed to check with OTA.