ayushsharma82/AsyncElegantOTA #97: [BUG] Crash before download completed

Alisofin, 2022-05-18:
[//]: <> (Please follow this template for fast resolutions)
**Describe the bug**
Hello,
Tested with ESP32-CAM both with Safari and  Chrome on MacBook. The upload stops quickly and the ESP32 resets. Please help.

<details>
11:15:09.472 -> abort() was called at PC 0x4016649d on core 1
11:15:09.472 -> 
11:15:09.472 -> Backtrace: 0x4009078c:0x3ffd1ab0 0x400909bd:0x3ffd1ad0 0x4016649d:0x3ffd1af0 0x4008b473:0x3ffd1b10 0x400e0531:0x3ffd1b30 0x400df86c:0x3ffd1b50 0x400dfa4b:0x3ffd1b70 0x400d275a:0x3ffd1b90 0x400d2899:0x3ffd1be0 0x400dd8b3:0x3ffd1c30 0x40178fc9:0x3ffd1c90 0x400d9c3d:0x3ffd1cc0 0x400db616:0x3ffd1d30 0x400db80d:0x3ffd1d70 0x400dea71:0x3ffd1d90 0x400deaf1:0x3ffd1dd0 0x400df18a:0x3ffd1df0 0x4008ced5:0x3ffd1e20
11:15:09.511 -> 
11:15:09.511 -> Rebooting...
11:15:09.511 -> ets Jun  8 2016 00:22:57
11:15:09.511 -> 
11:15:09.511 -> rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
11:15:09.511 -> configsip: 0, SPIWP:0xee
11:15:09.511 -> clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
11:15:09.511 -> mode:DIO, clock div:1
11:15:09.545 -> load:0x3fff0018,len:4
11:15:09.545 -> load:0x3fff001c,len:1216
11:15:09.545 -> ho 0 tail 12 room 4
11:15:09.545 -> load:0x40078000,len:9720
11:15:09.545 -> ho 0 tail 12 room 4
11:15:09.545 -> load:0x40080400,len:6352
11:15:09.545 -> entry 0x400806b8  

</details>

**Microcontroller Specs:**
 - Board: ESP32-CAM]
 - Microcontroller:  [ ESP32]
 - Did you modify the partition table? [No]
 - Browser: [chrome and safari]
 - Library Version: [latest]

**Additional context**
the AsyncWebServer works fine, server requested that I have implemented work fine.

Thanks for your help.

jackhax1, 2022-11-01:
Same issue

elpidiovaldez, 2023-01-15:
**Describe the bug**

Same issue.

The update stops very quickly, indicating 7% complete.  The output on the serial monitor implies that the system reset and reconnected to wifi.

Details

```
WiFi connected
Camera Ready and Over-The_Air updates available! Use 'http://192.168.1.156' to connect
ets Jun  8 2016 00:22:57

rst:0x8 (TG1WDT_SYS_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1344
load:0x40078000,len:13516
load:0x40080400,len:3604
entry 0x400805f0

.
WiFi connected
Camera Ready and Over-The_Air updates available! Use 'http://192.168.1.156' to connect

```

**Microcontroller Specs:**

Board: AI Thinker ESP32-CAM
Microcontroller: [ ESP32]
Did you modify the partition table? Using 'Minimal SPIFFS (1.9MB App with OTA/190KB SPIFFS)'
Browser: [chrome]
Library Version: name=AsyncElegantOTA, version=2.2.7

**Additional context**

AsyncWebServer works fine.

The main application is using two Async servers (one runs the ui and the other just does video streaming).  In principle one Async server could do both jobs, but I found the video frequently paused using a single web server.  I don't think this is the issue, because code update does not work even when I never start the video stream server.