ayushsharma82/AsyncElegantOTA #98: Alert Message: AsyncElegantOTA.loop()

limaomerces, 2022-05-19:
Using esp32 (with Arduino IDE) and compiling the example (Async_Demo), I got this ALERT:

```
In file included from C:\....\AppData\Local\Temp\arduino_modified_sketch_236353\Async_Demo.ino:19:
C:\....\Arduino\libraries\AsyncElegantOTA\src/AsyncElegantOTA.h:4:2: warning: #warning AsyncElegantOTA.loop(); is deprecated, please remove it from loop() if defined. This function will be removed in a future release. [-Wcpp]
 #warning AsyncElegantOTA.loop(); is deprecated, please remove it from loop() if defined. This function will be removed in a future release.
  ^~~~~~~
In file included from C:\.....\Arduino\libraries\AsyncElegantOTA\src\AsyncElegantOTA.cpp:1:
C:\.....\Arduino\libraries\AsyncElegantOTA\src/AsyncElegantOTA.h:4:2: warning: #warning AsyncElegantOTA.loop(); is deprecated, please remove it from loop() if defined. This function will be removed in a future release. [-Wcpp]
 #warning AsyncElegantOTA.loop(); is deprecated, please remove it from loop() if defined. This function will be removed in a future release.
  ^~~~~~~
```

In the author's example, there is no "AsyncElegantOTA.loop()" inside the "void loop(){}".

Should I be concerned? Did I miss something?

rubienr, 2022-06-23:
This warning [1] should be removed since it hinders clean coding because `-Wall -Werror` cannot be applied in `build_flags`.
Suggestion: remove the warning and if it breaks backward compatibility then bump the release version accordingly to indicate that fact.

Unfortunately I have to skip the usage of `ayushsharma82/AsyncElegantOTA@^2.2.7` as long its not compiling warning-free.

[1] 
https://github.com/ayushsharma82/AsyncElegantOTA/blob/0ad1b4af6a2f70b21fc4bfbc046d9930358b36fb/src/AsyncElegantOTA.h#L4

raselahamed613, 2022-08-27:
blutooth serial & AsyncElegantOTA dosent work togather. how to solved this problem?