ayushsharma82/AsyncElegantOTA #99: [BUG] Platformio & ES32

Eisbaeeer, 2022-05-29:
[//]: <> (Please follow this template for fast resolutions)
Error message after compiling/linking

**Do you get any error/exception stack? If so please paste it here:**
<details>
  <summary>Exception</summary>

  [//]: <> (Paste exception under this line)
  Linking .pio\build\esp32dev\firmware.elf
c:/users/lars weimar/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib1a0\libESP Async WebServer.a(WebAuthentication.cpp.o):(.literal._ZL6getMD5PhtPc+0x4): undefined reference to `mbedtls_md5_starts'
c:/users/lars weimar/.platformio/packages/toolchain-xtensa-esp32/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: .pio\build\esp32dev\lib1a0\libESP Async WebServer.a(WebAuthentication.cpp.o): in function `getMD5(unsigned char*, unsigned short, char*)':
C:\Users\Lars Weimar\Documents\GitHub\Balance-Robot_Kidbuild/.pio/libdeps/esp32dev/ESP Async WebServer/src/WebAuthentication.cpp:73: undefined reference to `mbedtls_md5_starts'
collect2.exe: error: ld returned 1 exit status
*** [.pio\build\esp32dev\firmware.elf] Error 1

</details>

**Microcontroller Specs:**
 - Board: [ ESP32 dev 1.0 ]
 - Microcontroller:  [ ESP32  ]
 - Did you modify the partition table? [No]
 - Browser: [chrome/safari/edge/opera]
 - Library Version: [latest]

**Additional context**
platformio.ini

[env:esp32dev]
platform = espressif32
board = esp32dev
framework = arduino
upload_speed = 921600
upload_port = COM3
monitor_port = COM3
monitor_speed = 115200
lib_deps = 
	ESP Async WebServer
	ayushsharma82/AsyncElegantOTA

JMartinezEco, 2022-06-20:
Hi! Any updates on this? I'm getting the same error.

RubenCarreira99, 2022-08-08:
@JMartinezEco I tried to paste that line of code onto platformio.ini and this is the error I get when compiling:

`Library Manager: Installing ayushsharma82/AsyncElegantOTA @ ^2.2.5 ottowinter/ESPAsyncWebServer-esphome @ ^2.1.0
Error: VCS: Unknown repository type ayushsharma82/AsyncElegantOTA @ ^2.2.5 ottowinter/ESPAsyncWebServer-esphome`

Eisbaeeer, 2022-08-10:
Worked for me, too.

closed 2022-08-10 by Eisbaeeer