ayushsharma82/AsyncElegantOTA #102: [BUG] Panic core_esp8266_main.cpp:137 __yield

emign, 2022-06-24:
**Describe the bug**
After the update has finished, the esp8266 (NodeMCU 12E) crashed with this exception

`User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:137 __yield

>>>stack>>>

ctx: sys
sp: 3ffffcf0 end: 3fffffb0 offset: 0000
3ffffcf0:  3ffffd1c 3ffe8ff8 3ffffd10 4020cfa4  
3ffffd00:  000000fe 00000000 00000000 00000000  
3ffffd10:  00000000 00000000 00000000 3fff3278  
3ffffd20:  00000000 00000000 00000008 4020cfa4  
3ffffd30:  3fff9544 3fff7d00 3fff8e5c 4021ffde  
3ffffd40:  3fff9544 00000001 3fff8e5c 4022004b  
3ffffd50:  3fff9544 00000020 3fff90cc 4021fab5  
3ffffd60:  3ffffd80 3fff7d00 3ffffd80 4021a145  
3ffffd70:  3fff9544 3fff7d00 3fff8e5c 4021a258  
3ffffd80:  00000000 001b001f 00000000 00000000  
3ffffd90:  3fff0065 00000000 3fff8aa0 4025e120  
3ffffda0:  00000001 2c9f0300 4000050c 000001df  
3ffffdb0:  000001df 3fff962a 3fff8e5c 4020e1d1  
3ffffdc0:  3fff8e5c 00000030 0000001c ffffffff  
3ffffdd0:  4025fcf4 0000ffff 000000e0 000001de  
3ffffde0:  000001df 3fff962a 3fff8e5c 4020c3a8  
3ffffdf0:  00000000 4bc6a7f0 83958106 0095c010  
3ffffe00:  00000000 00000000 4bc6a7f0 00000000  
3ffffe10:  000001de 00000001 401007d1 00000000  
3ffffe20:  00008183 3fff8cfc 3fff8a9c 3fff95e4  
3ffffe30:  3ffffe90 3fff8d74 3fff8a9c 4020c3fc  
3ffffe40:  3ffffe90 3fff8d74 3fff8a9c 40222df8  
3ffffe50:  000001df 3fff962a 3fff8d74 3fff8e5c  
3ffffe60:  00000000 00000000 3fff8a9c 4025a050  
3ffffe70:  3fff000a 3fff8a9c 3fff3ab5 00000000  
3ffffe80:  3fff95e4 3fff8a9c 3fff8d74 40222e58  
3ffffe90:  3fff8a58 3fff8a4c 00000001 3fff3664  
3ffffea0:  3fff38e4 3fff3664 3fff95e4 3fff3ab0  
3ffffeb0:  3fff8a9c 3fff3ab4 3fff3ab5 40260779  
3ffffec0:  00001000 3fffc6fc 0000005c 3fffdab0  
3ffffed0:  00000018 00000000 3ffe88c8 00000000  
3ffffee0:  3fff0000 00000215 00000020 3fff3664  
3ffffef0:  3fff38e4 3fff9602 3fff95e4 4025df01  
3fffff00:  00000014 3fff3664 3ffe88c8 40100f43  
3fffff10:  00000000 3fff5e14 3ffef188 3fff6bfc  
3fffff20:  3fffdc80 00000020 3fff8a34 3
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

0x4020cfa4 in AsyncWebServerResponse::addHeader(String const&, String const&) at ??:?
0x4020cfa4 in AsyncWebServerResponse::addHeader(String const&, String const&) at ??:?
0x4021ffde in raise_exception at core_esp8266_postmortem.cpp:?
0x4022004b in __panic_func at ??:?
0x4021fab5 in yield at ??:?
0x4021a145 in AsyncElegantOtaClass::restart() at ??:?
0x4021a258 in std::_Function_handler<void (AsyncWebServerRequest*), AsyncElegantOtaClass::begin(AsyncWebServer*, char const*, char const*)::{lambda(AsyncWebServerRequest*)#3}>::_M_invoke(std::_Any_data const&, AsyncWebServerRequest*&&) at AsyncElegantOTA.cpp:?
0x4025e120 in ip4_output_if_opt_src at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:1765
0x4020e1d1 in AsyncCallbackWebHandler::handleRequest(AsyncWebServerRequest*) at ??:?
0x4025fcf4 in tcp_input at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_in.c:153
0x4020c3a8 in AsyncWebServerRequest::_onData(void*, unsigned int) at ??:?
0x401007d1 in millis at ??:?
0x4020c3fc in std::_Function_handler<void (void*, AsyncClient*, void*, unsigned int), AsyncWebServerRequest::AsyncWebServerRequest(AsyncWebServer*, AsyncClient*)::{lambda(void*, AsyncClient*, void*, unsigned int)#8}>::_M_invoke(std::_Any_data const&, void*&&, AsyncClient*&&, std::_Any_data const&, unsigned int&&) at WebRequest.cpp:?
0x40222df8 in AsyncClient::_recv(std::shared_ptr<ACErrorTracker>&, tcp_pcb*, pbuf*, long) at ??:?
0x4025a050 in tcp_send_empty_ack at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_out.c:2058
0x40222e58 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, long) at ??:?
0x40260779 in tcp_input at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_in.c:501 (discriminator 1)
0x4025df01 in ip4_input at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:1467
0x40100f43 in umm_free_core at umm_malloc.cpp:?

[... 136 lines cut ...]
0x4025a050 in tcp_send_empty_ack at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_out.c:2058
0x40222e58 in AsyncClient::_s_recv(void*, tcp_pcb*, pbuf*, long) at ??:?
0x40260779 in tcp_input at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/lwip2-src/src/core/tcp_in.c:501 (discriminator 1)
0x4025df01 in ip4_input at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/lwip2-src/src/core/ipv4/ip4.c:1467
0x40100f43 in umm_free_core at umm_malloc.cpp:?
0x40256965 in ethernet_input_LWIP2 at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/lwip2-src/src/netif/ethernet.c:188
0x40256764 in git2glue_err at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:116
 (inlined by) esp2glue_ethernet_input at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/glue-lwip/lwip-git.c:477
0x402845e5 in ethernet_input at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:365
0x402845f7 in ethernet_input at /home/earle/src/esp-quick-toolchain/arduino/tools/sdk/lwip2/builder/glue-esp/lwip-esp.c:373
0x4027b7a6 in ppPeocessRxPktHdr at ??:?
0x4027fe37 in ets_snprintf at ??:?
0x40105aa5 in call_user_start_local at ??:?

--------------- CUT HERE FOR EXCEPTION DECODER ---------------`
  

</details>

**Microcontroller Specs:**
 - Board: NodeMCU 12 (Lolin)
 - Microcontroller:   ESP8266 
 - Did you modify the partition table? No
 - Browser: Chrome (using Safari the upload bar jumps to 100% and then nothing happens)
 - Library Version: latest

emign, 2022-06-27:
Damn

closed 2022-06-27 by emign

smeisner, 2023-04-17:
I just discovered that in AsyncElegantOTA.h (line 141, definition of the restart() func), it is calling yield() ...and delay() during a callback. I thought this is disallowed by AsyncWebServer! I commented out the call to restart() (line 86 in the AsyncElegantOTAClass definition) and everything works as expected. 

I do need to manually restart my ESP after upload completes. But no crash!

hmagarotto, 2023-04-19:
I think it's related to the issue: https://github.com/ayushsharma82/AsyncElegantOTA/issues/106 (this issue is still open)