ayushsharma82/AsyncElegantOTA #105: Improvment

simogaspa84, 2022-07-07:
Hi Mr. ayushsharma 82 ... 

I would like to know if you can adapt you lib for working also with ethernet cable.
The idea would be to use the same way but based on an ethernet webserver  based on this lib.. 
https://github.com/JAndrassy/EthernetENC

Please let me know if  are you going to do it.

I really appreciate.. 

Thanks

Sejny, 2022-08-14:
On esp32 its very easy. I use ElegantOTA over ethernet without problems.  Simply include ETH.h and configure PHY type, DHCP or static and enjoy. :)

simogaspa84, 2022-09-09:
Hi @Sejny can you provide an example of the code or a repo thank?

Sejny, 2022-09-09:
@simogaspa84   

Ethernet config is depend on used hardware.  

What hardware are you using?

simogaspa84, 2022-09-09:
enc28j60 ethernet shield

simogaspa84, 2022-09-12:
@Sejny can you provide the lines you changed ? Thanks a lot