ayushsharma82/AsyncElegantOTA #106: [BUG] Panic core_esp8266_main.cpp:137 __yield - OTA failed

mjurczak16, 2022-07-23:
[//]: <> (Please follow this template for fast resolutions)
**Describe the bug**
Cannot get OTA successful, hangs on 100% and MCU panics, other functionality works well.
Sample code for replication and screenshot attached. 
Developing on latest versions on `vscode` and `pio`

[image]

**Do you get any error/exception stack? If so please paste it here:**
<details>
  <summary>Exception</summary>

  [//]: <> (Paste exception under this line)
  Panic core_esp8266_main.cpp:137 __yield

>>>stack>>>

ctx: sys
sp: 3fffecb0 end: 3fffffb0 offset: 0000
3fffecb0:  3fff2374 0000001b 3fff239c 00000046  
3fffecc0:  000000fe 00000000 00000000 00000000  
3fffecd0:  00000000 00000000 00000000 3ffefa08  
3fffece0:  3fff0220 00000056 3ffe88c0 00000046  
3fffecf0:  40205a14 3fff1a04 3fff1fcc 4020fa4a  
3fffed00:  40205a14 00000020 3fff2374 4020fab7  
3fffed10:  3fffed3c 3fff1a04 3fffed3c 4020f2b5  
3fffed20:  40205a14 3fff1a04 3fff1fcc 40201e74  
3fffed30:  00000000 00000065 00000000 00000000  
3fffed40:  001b001f 00000000 3fff1e8c 402045c5  
3fffed50:  3fff1330 3fff1a04 3fff131c 40201e9c  
3fffed60:  3fff1e8c 3ffefde8 3fff131c 40206a44  
3fffed70:  3fff1a04 2c9f0300 4000050c 3fffc278  
3fffed80:  40103260 3fffc200 00000022 4010231c  
3fffed90:  3ffe8df2 00000030 00000010 ffffffff  
3fffeda0:  4022554d 00000000 00000014 00000046  
3fffedb0:  00000046 3fff22ca 3fff1a04 402050c8  
3fffedc0:  00000000 4bc6a7f0 0e560418 0511d758  
3fffedd0:  00000000 00000000 4bc6a7f0 00000000  
3fffede0:  00000045 00000001 40100599 00000000  
3fffedf0:  00046275 00004108 00000008 3fff2284  
3fffee00:  3fffee60 3fff18fc 3fff17c4 4020511c  
3fffee10:  3fffee60 3fff18fc 3fff17c4 40212178  
3fffee20:  00000046 3fff22ca 3fff18fc 3fff1a04  
3fffee30:  00000000 00000030 0000001a ffffffff  
3fffee40:  40240000 00000020 3ffef9ae 00000000  
3fffee50:  3fff2284 3fff17c4 3fff18fc 402121d8  
3fffee60:  3fff19f0 3fff19e4 3ffed2c0 3ffefde8  
3fffee70:  3fff0204 3ffefde8 3fff2284 3fff01d0  
3fffee80:  3fff17c4 3fff01d4 3fff01d5 40225f9d  
3fffee90:  3fff035c 0000007c 00000020 40100d58  
3fffeea0:  00000018 00000000 00000000 00000000  
3fffeeb0:  3fff0000 0000007c 00000020 3ffefde8  
3fffeec0:  3fff0204 3fff22a2 3fff2284 4022372d  
3fffeed0:  00000014 3ffefde8 3ffe88c0 40100b4f  
3fffeee0:  00000000 00000001 00000001 3fff0a7c  
3fffeef0:  3fffdc90 00000020 3fff1e8c 3fff0a7c  
3fffef00:  3ffefde8 00000008 3fff2284 4021c2a9  
3fffef10:  3fffdc90 3fff0604 3fff1e8c 4021c0a8  
3fffef20:  40249d65 3fff0604 3fff1e8c 40249d77  
3fffef30:  3fff2294 3fff2284 3fffdab0 00000000  
3fffef40:  4024102f 00000000 3fff1e8c 402455df  
3fffef50:  40000f49 3fffdab0 3fffdab0 40000f49  
3fffef60:  40000e19 00061f85 00000000 00000005  
3fffef70:  3fffefc0 aa55aa55 000000ed 401056b5  
3fffef80:  401056bb 00000000 00000005 40100d08  
3fffef90:  4010000d 736f4800 00061f85 401000ac  
3fffefa0:  40235fac 3fffef3c 40235f5d 3ffffdb8  
3fffefb0:  3fffffc0 00000000 00000000 feefeffe  
3fffefc0:  feefeffe feefeffe feefeffe feefeffe  
3fffefd0:  feefeffe feefeffe feefeffe feefeffe  
[... 269 lines cut ...]

void setup() {

  ESP.wdtDisable();
  ESP.wdtEnable(30000);

  WiFi.softAPIP();
  AsyncWebServer server(80);

  WiFi.softAP("ESP8266_mytest","passwd");

  server.on("/", HTTP_GET, [](AsyncWebServerRequest *request) {
      request->send(200, "text/plain", "Hi! I am ESP32.");
    });

  AsyncElegantOTA.begin(&server); // Start ElegantOTA
  server.onNotFound(notFound);
  server.begin();

  while (1) {
    ESP.wdtFeed();
    delay(100);
  }
}
```

hervema, 2022-12-28:
Hello. I got exactly the same problem as mjurczak16. Is there any news of this issue ? 

Thanks a lot.

jitendra1607, 2023-03-01:
PLATFORM: Espressif 8266 (4.0.1) > NodeMCU 1.0 (ESP-12E Module)
HARDWARE: ESP8266 160MHz, 80KB RAM, 4MB Flash

i am also getting same error on above module.

`
User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:137 __yield

>>>stack>>>

ctx: sys
sp: 3fffeb80 end: 3fffffb0 offset: 0000
3fffeb80:  3ffeace0 2c9f0300 4000050c 3fffc278  
3fffeb90:  000000fe 00000000 00000000 00000000  
3fffeba0:  00000000 00000000 00000000 3fff4e8c  
3fffebb0:  aaaaaaaa aaaaaaaa 3ffede98 00000000  
3fffebc0:  00000218 00000001 3fff4d44 4021fad2  
3fffebd0:  00007fff 0bdbc4a1 3ffee7b4 4021fb3f  
3fffebe0:  3ffeacec 00000000 00000000 4021f5a9  
3fffebf0:  00007fff 0bdbc4a1 40104002 4021dd2d  
3fffec00:  3ffeacec 7fffffff 00002200 00000001  
3fffec10:  00000001 00004288 3fff829c 00000000  
3fffec20:  00000218 00000218 3fff4d44 4021de6c  
3fffec30:  3fff929c 00000054 00000fac 00001000  
3fffec40:  40103d24 3fffecd0 3fff8764 00000000  
3fffec50:  00000fac 00000218 3fff8764 4020558f  
3fffec60:  4020e701 000005b4 00000156 00000000  
3fffec70:  00000155 00000156 6d726966 65726177  
3fffec80:  08000000 00000050 00000000 00000000  
3fffec90:  000000c1 00001215 3fff8764 00000030  
3fffeca0:  3ffe8d58 4010605f 3ffedba0 3fff70d0  
3fffecb0:  3fff70bc 3fff8848 3fff8764 40210834  
3fffecc0:  3fffece0 3fff9034 3fff9034 4022e49b  
3fffecd0:  00000218 3fff929c 00000fac 3fff8764  
3fffece0:  3fff5600 00000048 00000005 40102de4  
3fffecf0:  00000fac 3fff929c 00000218 00000000  
3fffed00:  3ffe8d58 3ffede70 3fff5638 40234508  
3fffed10:  00007fff 0bdbc34f 3ffee7b4 3fff8764  
3fffed20:  000012d7 00000038 3fff8764 40222d78  
3fffed30:  00000000 0bdbc34f 40104002 3fff8764  
3fffed40:  00001297 000000f0 00000001 4020d6c0  
3fffed50:  00000001 00004288 00000062 693df325  
3fffed60:  3ffeace0 3fff5310 3fff67a0 0bdbc34f  
3fffed70:  3ffeacec 2c9f0300 4000050c 3fffc278  
3fffed80:  40103d24 3fffc200 00000022 00000000  
3fffed90:  40237afa 00000030 00000010 ffffffff  
3fffeda0:  40237adc 0000ffff 000000e0 00000217  
3fffedb0:  00000218 3fff907a 3fff8764 4020e701  
3fffedc0:  00000000 4bc6a7f0 fb22d0e5 0382963a  
3fffedd0:  00000000 00000000 4bc6a7f0 00000000  
3fffede0:  00000217 00000001 401010ad 00000000  
3fffedf0:  00030927 00000047 00000002 3fff9034  
3fffee00:  3fffee60 3fff867c 3fff679c 4020e888  
3fffee10:  3fffee60 3fff867c 3fff679c 4022231c  
3fffee20:  00000218 3fff907a 3fff867c 3fff8764  
3fffee30:  00000000 2c9f0300 4000050c 3fffc278  
3fffee40:  40100000 3fffc200 00000022 00000000  
3fffee50:  3fff903
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:137 __yield

>>>stack>>>

ctx: sys
sp: 3fffe8b0 end: 3fffffb0 offset: 0000
3fffe8b0:  4026c195 00000000 3fffe910 40226620  
[... 70 lines cut ...]
3fffed20:  000012d7 00000038 3fff8764 40222d78  
3fffed30:  00000000 0bdbc34f 40104002 3fff8764  
3fffed40:  00001297 000000f0 00000001 4020d6c0  
3fffed50:  00000001 00004288 00000062 693df325  
3fffed60:  3ffeace0 3fff5310 3fff67a0 0bdbc34f  
3fffed70:  3ffeacec 2c9f0300 4000050c 3fffc278  
3fffed80:  40103d24 3fffc200 00000022 00000000  
3fffed90:  40237afa 00000030 00000010 ffffffff  
3fffeda0:  40237adc 0000ffff 000000e0 00000217  
3fffedb0:  00000218 3fff907a 3fff8764 4020e701  
3fffedc0:  00000000 4bc6a7f0 fb22d0e5 0382963a  
3fffedd0:  00000000 00000000 4bc6a7f0 00000000  
3fffede0:  00000217 00000001 401010ad 00000000  
3fffedf0:  00030927 00000047 00000002 3fff9034  
3fffee00:  3fffee60 3fff867c 3fff679c 4020e888  
3fffee10:  3fffee60 3fff867c 3fff679c 4022231c  
3fffee20:  00000218 3fff907a 3fff867c 3fff8764  
3fffee30:  00000000 2c9f0300 4000050c 3fffc278  
3fffee40:  40100000 3fffc200 00000022 00000000  
3fffee50:  3fff9034 3fff679c 3fff867c 4022237c  
3fffee60:  3fff6758 3fff674c 68f2bae2 3fff5310  
3fffee70:  3fff5590 3fff5310 3fff9034 3fff575c  
3fffee80:  3fff679c 3fff5760 3fff5761 40238561  
3fffee90:  00000000 3fffdad0 3fff4e44 00000030  
3fffeea0:  00000010 00000000 3ffe8614 00000000`

jitendra1607, 2023-03-02:
@ayushsharma82 any solution?

hmagarotto, 2023-04-19:
On "randomnerdtutorials" they "schedule" the restart performed on the main loop:

https://randomnerdtutorials.com/esp8266-nodemcu-wi-fi-manager-asyncwebserver/

@smeisner disables the restart completely as a workaround: https://github.com/ayushsharma82/AsyncElegantOTA/issues/102#issuecomment-1511835062

I hope it helps.

JoJos1220, 2023-05-07:
I had exactly the same issue that @jitendra1607 had described within the Post above!
I Did it like described and it helped:

_just discovered that in AsyncElegantOTA.h (line 141, definition of the restart() func), it is calling yield() ...and delay() during a callback. I thought this is disallowed by AsyncWebServer! I commented out the call to restart() (line 86 in the AsyncElegantOTAClass definition) and everything works as expected.

I do need to manually restart my ESP after upload completes. But no crash!_

Thank you @smeisner for that solution it helps temporairly! Hopefully the libary will get soon updated to prevent that issue in long term....

Regards!

jitendra1607, 2023-05-09:
@JoJos1220 I tried your suggestion. even after rebuild and upload same error is occurring as previous. I think i am facing this issue because of memory partition.

JoJos1220, 2023-05-09:
Ok, this is strange....do you get definetly 1:1 the same error message? Can you also check the result after your build in platformio the amount of ram/Flash needed?
I don't know if the littleFS Partition also affects this error message....

Regards!

jitendra1607, 2023-05-10:
```

User exception (panic/abort/assert)
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:137 __yield

>>>stack>>>

ctx: sys
sp: 3fffeb80 end: 3fffffb0 offset: 0000
3fffeb80:  40205520 00000050 00000000 3fff867c  
3fffeb90:  000000fe 00000000 00000000 00000000  
3fffeba0:  00000000 00000000 00000000 3fff49bc  
3fffebb0:  aaaaaaaa aaaaaaaa 3ffed990 00000000  
3fffebc0:  00000218 00000001 3fff4874 40221772  
3fffebd0:  00007fff 05301275 3ffee2ec 402217df  
3fffebe0:  3ffea7dc 00000000 00000000 40221249  
3fffebf0:  00007fff 05301275 40256923 4021f739  
3fffec00:  00000002 00000000 00000020 40100898  
3fffec10:  40103211 3ffed6e8 00000002 00000000  
3fffec20:  00000218 00000218 3fff4874 4021f878  
3fffec30:  3fff8e5c 00000054 00000fac 00001000  
3fffec40:  401056ab 3fffecd0 3fff867c 00000000  
3fffec50:  00000fac 00000218 3fff867c 40205593  
3fffec60:  40105173 0000002d 00000002 00040000  
3fffec70:  00002200 401066bb 6d726966 65726177  
3fffec80:  08000000 00080000 00000020 40100898  
3fffec90:  40103211 00000000 3fff737c 3fff4e28  
3fffeca0:  3ffe8c8a 2c9f0300 4000050c 3fff7b38  
3fffecb0:  3fff7b24 3fff8760 3fff867c 40210fbc  
3fffecc0:  3fffece0 00000030 00000010 ffffffff  
3fffecd0:  00000218 3fff8e5c 00000fac 3fff867c  
3fffece0:  00000000 00000026 00000648 00000430  
3fffecf0:  00000fac 3fff8e5c 00000218 00000000  
3fffed00:  3ffe8c8a 00000218 00000025 00000030  
3fffed10:  00007fff 00000000 3ffee2ec 3fff867c  
3fffed20:  000012d7 00000038 3fff867c 40224c68  
3fffed30:  00000000 00040000 00000000 3fff867c  
3fffed40:  00001297 000000f0 00000001 4020de48  
3fffed50:  3ffee118 40103ed3 00001043 00000020  
3fffed60:  3fff8418 00000000 3fff80d8 402380a4  
3fffed70:  00000218 2c9f0300 4000050c 3fffc278  
3fffed80:  40103d24 3fffc200 00000022 00000000  
3fffed90:  4023a69c 00000030 0000001c ffffffff  
3fffeda0:  4023a45e 00000000 00000000 00000217  
3fffedb0:  00000218 3fff8c3a 3fff867c 4020ee89  
3fffedc0:  00000000 4bc6a7f0 a1cac083 01891e6b  
3fffedd0:  00000000 00000000 4bc6a7f0 00000000  
3fffede0:  00000217 00000001 401010ad 00000000  
3fffedf0:  000153fe 00000030 0000001c 3fff8bf4  
3fffee00:  3fffee60 3fff8594 3fff80d4 4020f010  
3fffee10:  3fffee60 3fff8594 3fff80d4 40223fbc  
3fffee20:  00000218 3fff8c3a 3fff8594 3fff867c  
3fffee30:  00000000 3fffdc80 00000000 00000030  
3fffee40:  3fff0000 3fff80d4 3fff5291 00000000  
3fffee50:  3fff8bf4 3fff80d4 3fff8594 4022401c  
3fffee60:  3fff7e1
--------------- CUT HERE FOR EXCEPTION DECODER ---------------

Panic core_esp8266_main.cpp:137 __yield

>>>stack>>>

ctx: sys
sp: 3fffe8b0 end: 3fffffb0 offset: 0000
3fffe8b0:  4026e3cd 00000000 3fffe910 40228610  
3fffe8c0:  000000fe 00000000 00000000 00000000  
3fffe8d0:  00000000 00000000 00000000 4022cca4  
3fffe8e0:  00000000 4026e3cd 00000001 00002200  
3fffe8f0:  3fffc200 40103d24 04000000 40102702  
3fffe900:  00000003 00000004 40003b53 00000000  
3fffe910:  40100fff 00000000 00000000 40103d0a  
3fffe920:  40103d81 00000031 40302063 00000000  
3fffe930:  40100fff 00000000 00000000 00000004  
3fffe940:  40003b53 2c9f0300 4000050c 3fffc278  
3fffe950:  40103d24 3fffc200 00000022 4022cca4  
3fffe960:  40003b53 00000030 00000010 ffffffff  
3fffe970:  40001ddc 00000030 00018000 00000000  
3fffe980:  00ff0000 5ffffe00 5ffffe00 00000000  
3fffe990:  00018000 00000000 3fffe953 3fffea83  
3fffe9a0:  00000020 0000003c 3fffffb0 00000030  
3fffe9b0:  0000003c 3ffe8368 000000a0 4022a349  
3fffe9c0:  3fffeaa1 00000001 0000006e ffff0208  
3fffe9d0:  3fffea70 0000009f 3fffefc0 3fffefc0  
3fffe9e0:  3fffffb0 3ffe8368 000000a0 4022a349  
3fffe9f0:  3fffeac8 0000009f 00000077 ffff0208  
3fffea00:  3fffeaa0 0000009f 40104002 00000100  
3fffea10:  3ffea7dc 7fffffff 00002200 00000001  
3fffea20:  00000001 00000080 00000000 00000030  
3fffea30:  3ffea7dc 00000020 3fffea70 4022a385  
3fffea40:  3fffeb30 3fffeb20 00000004 3fffc278  
3fffea50:  40103d24 3fffeb80 3fffea90 4022a385  
3fffea60:  40221455 3fffeb80 3fffeaa0 40221448  
3fffea70:  66666633 30366565 3320203a 37666666  
3fffea80:  20303165 66666633 34306537 66663320  
3fffea90:  37393466 66332034 65346666 20203034  
3fffeaa0:  203a000a 66666633 30386265 646e6520  
3fffeab0:  6633203a 66666666 6f203062 65736666  
3fffeac0:  30203a74 0a303030 00000000 00000001  
3fffead0:  00000001 00004288 00000000 3fffec40  
3fffeae0:  40221455 00000000 00000020 40221448  
3fffeaf0:  6573550a 78652072 74706563 206e6f69  
3fffeb00:  6e617028 612f6369 74726f62 7373612f  
3fffeb10:  3fffeb30 3fffeb20 00000004 ffffffff  
3fffeb20:  00000001 3fffee60 3fff7e10 3fff7e04  
3fffeb30:  3fff4974 3fff4e40 00000004 053018b7  
3fffeb40:  3fffeb60 3fffeb50 3fffee70 402214a2  
3fffeb50:  00000020 3fffeb80 3fffffb0 00000000  
3fffeb60:  5ffffe00 5ffffe00 0000000f 3fffefc0  
3fffeb70:  3fffffb0 3fffeb80 3fffeb80 40221702 <
3fffeb80:  40205520 00000050 00000000 3fff867c  
3fffeb90:  000000fe 00000000 00000000 00000000  
3fffeba0:  00000000 00000000 00000000 3fff49bc  
3fffebb0:  aaaaaaaa aaaaaaaa 3ffed990 00000000  
3fffebc0:  00000218 00000001 3fff4874 40221772  
3fffebd0:  00007fff 05301275 3ffee2ec 402217df  
3fffebe0:  3ffea7dc 00000000 00000000 40221249  
3fffebf0:  00007fff 05301275 40256923 4021f739  
3fffec00:  00000002 00000000 00000020 40100898  
3fffec10:  40103211 3ffed6e8 00000002 00000000  
3fffec20:  00000218 00000218 3fff4874 4021f878  
3fffec30:  3fff8e5c 00000054 00000fac 00001000  
3fffec40:  401056ab 3fffecd0 3fff867c 00000000  
3fffec50:  00000fac 00000218 3fff867c 40205593  

```