ayushsharma82/AsyncElegantOTA #112: 401 error on platformio_upload.py script when uploading

RFlintstone, 2022-10-01:
[//]: <> (Please follow this template for fast resolutions)
**Describe the bug**
I'm getting a forbidden ``<Response [401]>`` message when trying to automatically upload to the OTA service which has a username and password on. Is there a way to fix/bypass this with the ``platformio_upload.py`` script?

**Do you get any error/exception stack? If so please paste it here:**
<details>
  <summary>Exception</summary>

  [//]: <> (Paste exception under this line)
  <Response [401]>

</details>

**Microcontroller Specs:**
 - Board: firebeetle32
 - Microcontroller:  ESP32(-e) from DFRobot
 - Did you modify the partition table? [Yes/**No**]
 - Browser: [chrome/safari/edge/opera/**brave (edited chromium)**]
 - Library Version: latest

**Additional context**
Besides this issue, everything seems to work fine.

mmrvelj, 2022-11-01:
Until somebody does a better job fix, here is a quick help for this.

In the platformio_upload.py script replace line 49:
` response = requests.post(upload_url, data=monitor, headers={'Content-Type': monitor.content_type})`
with
` response = requests.post(upload_url, auth=('user', 'password'),data=monitor, headers={'Content-Type': monitor.content_type})`

of course, replacing **user** and **password** with values that work for you.

TheXtremeKing, 2023-08-06:
> Until somebody does a better job fix, here is a quick help for this.
>
> In the platformio_upload.py script replace line 49: ` response = requests.post(upload_url, data=monitor, [...]

Thanks, this worked for me!