ayushsharma82/AsyncElegantOTA #125: [BUG]: Compilation error: ESPAsyncWebServer.h: No such file or directory - with possible resolution

juhulu, 2023-03-11:
[//]: <> 
First of all thanks for the great work on ElegantOTA!
I have started using your OTA library just recently, and I ran into an issue that might be of general interest (or just a mere sign of my ineptitude, in this case please gently ignore): 
When I compiled some code under the Arduino IDE (ver. 2.0.4, Windows 11, 64 Bit), I received the following error:

> In file included from [xxxxx].ino:69:
> c:\Users\[xxxxx]\Arduino\libraries\AsyncElegantOTA\src/AsyncElegantOTA.h:24:10: fatal error: ESPAsyncWebServer.h: No such file or directory
>  #include "ESPAsyncWebServer.h"
>           ^~~~~~~~~~~~~~~~~~~~~
> compilation terminated.
> 

However, the respective library (ESPAsyncWebServer) is installed (version 1.2.6 by dvarrel).

After some searching, the following solved the issue for me: 
Replacing line 23 in AsyncElegantOTA.h (version 2.2.9) as follows
old: #include "ESPAsyncWebServer.h"
new: #include "ESPAsyncWebSrv.h"
works like a breeze.

As I said: maybe a stupid glitch I caused in my installation, but maybe of interest to others.

**Microcontroller Specs:**
 - Board: WEMOS D1 MINI ESP32
 - Microcontroller:  ESP32
 - Did you modify the partition table? No
 - Browser: edge
 - Library Version: 2.2.9

Jaybee2020, 2023-03-23:
The reason is that ESPAsyncWebServer and ESPAsyncWebSrv are two different libraries. AsyncElegantOTA expects the first one found here:

https://github.com/me-no-dev/ESPAsyncWebServer

ACure4IT, 2023-03-24:
I created this pull request which should resolve this issue.

https://github.com/ayushsharma82/AsyncElegantOTA/pull/129