ayushsharma82/AsyncElegantOTA #128: [BUG] AsyncElegantOTA 2.2.7 does not show success message

Jaybee2020, 2023-03-23:
[//]: <> (Please follow this template for fast resolutions)
**Describe the bug**
After succesfully uploading a firmware image, no success message is shown. 

**Do you get any error/exception stack? If so please paste it here:**
<details>
  <summary>Exception</summary>

  [//]: <> (Paste exception under this line)
  

</details>

**Microcontroller Specs:**
 - Board: Wemos D1 Mini
 - Microcontroller:  [ ESP8266 ]
 - Did you modify the partition table? [No]
 - Browser: [chrome]
 - Library Version: [latest]

**Additional context**
Add any other context about the problem here.

DONGILBAE, 2023-06-18:
me too. when updating, stop process and reboot. 
ESP32-U
arduino ide 2.0.4

Jaybee2020, 2023-06-19:
The solution is ease. You have to skip the internal reset procedure of AsyncElegantOTA. This procedure contains a yield which blocks the webserver. Instead set a flag when update is finished and do the reset from main loop.

Daryavahus, 2023-07-23:
I changed yield() to esp_yield(), but it doesn't change anything.  I've noticed that when downloading has finished ESP lost ip address. So connecting to Asyncwebserver is not possible.
And now question is:
What should I do to prevent WiFi disconnecting?

Daryavahus, 2023-08-03:
Thank You. Your're. It works. I checked earler versions and library has esp reset in loop void. Then it was moved to reset void.

dreamy1, 2023-08-05:
Same problem here :-)

Just gone back to EOTA 2.26, but with no success. The update itself works fine, but the webserver stays at 100% with no feedback of success.

Daryavahus, 2023-08-05:
I changed asyncelegantota.cpp file. Find restart() void. Remove or comment all inside.
Add 
static bool _resetesp = false;
In first line of cpp file.
Add code
_resetesp = true;
In restart() void
And in some place In loop code:
If(_resetesp){
//ESP.reset();
ESP.restart();
//depend what You need to do ucomment first or second line
}

Try this and Just let us know.

dreamy1, 2023-08-05:
Thanks a lot!

Before i manually change parts of the lib...do you think there will be an update of the whole lib in future? It seems that the last revision is more than one year old...and i love this lib :-)

Daryavahus, 2023-08-06:
I change it in my project and it works fine. Restart() void is called as the last part of library after downloaded firmware and copied to flash, so if it doesn't work, Your ESP will not restart. Look at the older version of AEOTA(1.0 or above).