ayushsharma82/AsyncElegantOTA #139: AsyncElegantOTA through URL call with file name

MikeF5J, 2023-07-28:
Ayush,
The ElegantOTA works well. I would like to try and simplify the call to the device and do the FW update through an auto created URL.

Can you assist by helping with such a URL construct?
e.g.
ip/update/ x??x "fw filename & path" + button submitted

By sending the URL to the browser the IP should be started and the fw file selected and added and start the OTA automatically.

The coffees I'll save I'll pass to you.

Thank you
Mike

nozbey, 2023-09-06:
Do you need http get request, instead of post method?

MikeF5J, 2023-09-06:
Any method can probably work.  If you know how post method works please let me know.  I just want to create the URL construct in code - I do not want to type it in a browser or select files from a button or entry field.  And making (get or post) the URL call the  OTA process must start automatically with the file specified.  No browser screen or UI is required then.