ayushsharma82/AsyncElegantOTA #142: Problem uploading firmware (bad http request) [BUG]

VALERYRAMIREZ, 2023-08-20:
Hi all, how are you?

I'm using several esp-01s  and I'm able to upload to file system memory on everyone of them but, when I'm trying to upload firmware then in some of them the progress bar shows 100% and after that it shows an error on the web page that says something like http bad request, on others it loads around 50% and after that ESP device resets itself without loading the new code and the web page of asyncelegantOTA crashes.

It is for sure that I'm uploading the code selecting the firmware checkbox and the filesystem info by selecting the filesystem checkbox.

How can this be solved?

tanasecosminromeo, 2023-09-20:
same device, same issue

tanasecosminromeo, 2023-09-20:
@VALERYRAMIREZ I am doing some progress here so check back when u can

In src/AsyncElegantOTA.cpp I've commented out restart(); - this kind of did the trick as in the upload is successful.

Since the sketch was too big for 1MB I have initially tried to use this ldscript
```
board_build.ldscript = eagle.flash.1m64.ld
```

Success!!... not. Without the restart and with that ldscript both the serial (i've added some serial prints for debugging) returned everything as successful, but.. no luck, when I have manually restarted my old sketch was still there.

Ok.. I then seen here that I could compress the firmware https://arduino-esp8266.readthedocs.io/en/latest/ota_updates/readme.html#compression .. but after the Update was complete...

```
User exception (panic/abort/assert)
Panic core_esp8266_main.cpp:191 __yield
```

@ayushsharma82 Any tips? I believe it's something to do with 
```
                size_t fsSize = ((size_t) &_FS_end - (size_t) &_FS_start);
                uint32_t maxSketchSpace = (ESP.getFreeSketchSpace() - 0x1000) & 0xFFFFF000;

                if (!Update.begin((cmd == U_FS)?fsSize:maxSketchSpace, cmd)){ // Start with max available siz
```

Sorry to bother you and GG for this library, worked great for my ESP32

ayushsharma82, 2023-09-20:
@VALERYRAMIREZ  Can you give ElegantOTA V3 a try and see if the problem still persists?  This repo will soon be deprecated as I'm merging async functionality in original library.

Link to V3: https://github.com/ayushsharma82/ElegantOTA/tree/v3
Documentation for enabling async mode: https://docs.elegantota.pro/async-mode/

ayushsharma82, 2023-09-20:
@tanasecosminromeo You are invited to join the beta run of V3 too 😄